prusa3d/PrusaSlicer #8344: Gcode Viewer error

sisuwest, 2022-05-21:
### Description of the bug

I can't open the gcode viewer window unless PrusaSlicer is already running.
[image]
 

### Project file & How to reproduce

Try to open any gcode and you'll get that error. 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+arm64

### Operating system

macOS 12.3

### Printer model

mk3s+

rtyr, 2022-05-21:
Already reported here https://github.com/prusa3d/PrusaSlicer/issues/8300. I am closing this one as a duplicate.

closed 2022-05-21 by rtyr

bubnikv, 2022-06-29:
When trying to open a G-Code file using drag and drop or "Open with" option, G-Code Viewer would report the file extension as unknown when PrusaSlicer was not already running.

Fixed in PrusaSlicer 2.5.0-alpha2.

katomic, 2022-09-15:
I also encountered this issue (https://forum.prusa3d.com/forum/prusaslicer/prusaslicer-2-4-2arm64-gcode-viewer-error-on-macos-on-first-launch/) on a MBP M1.   Kudos to the developer/s for fixing this!