prusa3d/PrusaSlicer #8345: new infill for empty with small ribs

psychotik2k3, 2022-05-22:
Hi i just watched this video where the youtuber explain how he removes small 00.2mm width ribs from his model so when he prints in vase mode these ribs will generate a hollow model but with structure so it's more rigid

https://www.youtube.com/watch?v=STNJbHJf9K0

I think these would be best for an option in the slicer, like how many ribs you want

[image]
[image]
[image]

foreachthing, 2022-05-22:
I never use vase mode because the models are flimsy. But with this .....!! 😮

bubnikv, 2023-06-19:
You need to decrease the slicing closing radius, similar to 
https://github.com/prusa3d/PrusaSlicer/issues/4245

closed 2026-09-29 by prusaslicer-bot[bot]