prusa3d/PrusaSlicer #8347: Draft shield has no brim

espr14, 2022-05-22:
### Description of the bug

Some filaments have high tendency to warp after extrusion so a brim is used to keep the first layer attached to the bed. Shield blocks the cool air so the warping is smaller. But the shield has no brim so the shield can detach during printing. This happens if object's first layer is much smaller than the object in the upper layers or if the object is tilted.

### Project file & How to reproduce

1. generate a cube
2. rotate it 45° in X (cube will stand on the edge)
3. turn brim on
4. turn skirt on
5. switch draft shield to Enabled
6. switch to preview screen and render the model

Cube has brim but draft shield has not.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+linux-x64-GTK3

### Operating system

Linux Mint 20.2 Cinnamon

### Printer model

Prusa i3

Gregor-Gregor, 2022-08-23:
Draft shied usually helps with draft vulnerable filaments. But this also means the draft shield itself is vulnerable, may warp and detach, unless number of loops is increased... increasing also waste of filament to construct unnecessarily thick draft shield.
Brim to the draft shield - yes, please.

jbernardis, 2022-09-05:
I too vote for this feature.  If I'm having enough trouble with adhesion that I need to use a draft shield, then it makes sense that that shield will aldo have trouble with adhesion.  A brim on the shield will mitigate this.

n8bot, 2022-09-05:
I use the draft shield to purge tool changes on my IDEX, but I also needed a brim for the skirt so I have an open PR to include this: https://github.com/prusa3d/PrusaSlicer/pull/7305

subatomicglue, 2022-10-04:
I'd love to print the part without a brim, but with a draft shield that has a brim.  makes total sense.   +1

**Additional Feature Request** - draft shield should not only be convex hull as it is now, but should follow the same path as the brim for the tightest draft shield possible!   (I suppose it could be nice to select which).   Often there are times when the draft shield is quite far away from some interior feature of what I'm trying to shield, kind of defeating the purpose.

Illustrated, I hope you see the benefit of a concave hull for the draft shield!:
[image]

Gregor-Gregor, 2022-10-04:
Good idea, _subatomicglue_. 👍

I was just thinking about similar concept of a **3D draft shield**, following the shape of the object all along Z-axis as well, like a glove (with all its concavities).

There are several issues of such a _3D draft shield_ to be taken into consideration, eg.:
- maximum slope angle to avoid overhangs
- following shape of the supports as well
- concavities smoothing radius to avoid following all the tiny details of the object when distance from it is set small
- how to remove it later without shears 😉.

Nonetheless, such a _3D draft shield_ (brimmed at the bottom, of course) should allow using even most draft sensitive materials _without_ an enclosure.

Cheers

subatomicglue, 2022-10-04:
Really good idea.

Order of work that seems smart, IMHO:
- **Draft Shield Brim**:   add brim to existing draft shield (for both brimless parts, and parts with brims)
- **Concave Draft Shield**:  allow a concave "tight fit" draft shield instead of the current convex one
- **3D Draft Shield**:   explore the 3D draft shield concept, for using even most draft sensitive materials without an enclosure.

Each of these increments would (likely) build on the previous, and provide incremental value to users...  And the 3D one feels like some R&D (harder, more to research/discover), if I had to guess.

GMeyerGMeyer, 2023-05-07:
Would love to have it too.

- draft shield follows Object form
- Draft shield gets an option to create its own outer and inner brim. (my shield always develops its own will... )

closed 2026-09-29 by prusaslicer-bot[bot]