prusa3d/PrusaSlicer #8348: More Options for Custom GCODE in the "Filament Settings" Tab

sfrayy, 2022-05-23:
**Is your feature request related to a problem? Please describe.**
I need custom GCODE at the end of each layer depending on the filament type, because my metal filament needs a nozzle clean routine which other filaments don't need.

**Describe the solution you'd like**
If the custom GCODE option for "GCODE after layer change" shift would not only be available in the "Printer Settings" tab but also in the "Filament settings" tab, my problem would be fixed. At the moment the custom GCODE in "Filament Settings" only includes GCODE at start and end of a print.  

**Describe how it would work**
Just make all options for custom GCODE available on the "Filament Settings" tab as well, everything is already implemented. (should be a simple change.)  

**Describe alternatives you've considered**
The alternative is creating a custom Printer preset. This is a perfect workaround, but requires good explaining when the printer is used by multiple people, as you need to combine different filaments with different printer settings, which should not be the case.

**Additional context**
I am trying to modify an i3 MK3 Printer to print metal filament from another supplier which has a very rudimentary slicer that only works for their locked down system of slicer and printer. To get more options on print settings I need to transfer their settings to the MK3. Therefore I mounted the Auto-Nozzle-Cleaner from the printables library, as the original printer also has a nozzle cleaner. This cleaning routine should only be triggered for certain filaments and not for all. Thank you in advance.

stepikovo, 2022-05-26:
You can, as a temporary solution, use macros, for example `{if filament_settings_id[0]=="ColorFabb bronzeFill"}your macro here{endif}`. This then could be your default Printer profile, because the cleaning routine will be active only with the one specific filament.

sfrayy, 2022-05-30:
Thank you very much for your help @stepikovo. This is a perfect solution for me. I'll still leave this request open, as more options in the "Filament Settings" tab would be the more logical way.

closed 2026-09-29 by prusaslicer-bot[bot]