prusa3d/PrusaSlicer #8349: New Variable in custom GCODE: do_for_every_x_layer

sfrayy, 2022-05-23:
**Is your feature request related to a problem? Please describe.**
I want to trigger certain GCODE at the end of every third layer. 

**Describe the solution you'd like**
I would like to be able to use the if-function to execute a custom GCODE at every X layer.  

**Describe how it would work**
I suggest a new variable for the "custom GCODE" option, where you can trigger a custom GCODE every X layer. 

**Describe alternatives you've considered**
The alternative would be to either trigger the GCODE at every single layer or to manually insert it after slicing at the correct position. The first alternative is simple, but enlarges the print time, the second enlarges the time needed to post-process the GCODE after slicing.

**Additional context**
I am trying to modify an i3 MK3 Printer to print metal filament from another supplier which has a very rudimentary slicer that only works for their locked down system of slicer and printer. To get more options on print settings I need to transfer their settings to the MK3. Therefore I mounted the Auto-Nozzle-Cleaner from the printables library, as the original printer also has a nozzle cleaner. It's cleaning routine is executed on every third layer. I would like to implement someting similar. Thank you in advance.

bubnikv, 2022-05-23:
you may implement it yourself with if block and modulo operator.

see
Macros | Prusa Knowledge Base (prusa3d.com)
<https://help.prusa3d.com/article/macros_1775>

sfrayy, 2022-05-23:
@bubnikv Thank you. I didn't find that page. After a bit of headache I finally found a solution to the problem:

{if round(layer_num / 3) * 3 == layer_num}
;do something
{endif}

So what this does is divide the layer number by 3, then round the result and multiply it by 3 again. The result is then compared to the actual layer_num again. Only if layer_num is a multiple of 3, the equation will be valid. 

As I found a solution to this, I will close this request now. TY!

closed 2022-05-23 by sfrayy