prusa3d/PrusaSlicer #8350: Ubuntu 22.04 FUSE2 support

paulscott56, 2022-05-23:
### Description of the bug

When attempting to run PrusaSlicer (PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056.AppImage) on Ubuntu 22.04 or later, you will get an error message 

dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information

This is caused by fuse2 not being installed on Ubuntu 22.04 and later and rather fuse3 being installed.

In order to fix this, first check if fuse3 is installed

dpkg -l | grep fuse3
ii  fuse3                                      3.10.5-1build1                          amd64        Filesystem in Userspace (3.x version)
ii  libfuse3-3:amd64                           3.10.5-1build1                          amd64        Filesystem in Userspace (library) (3.x version)

If it is, you may install libfuse2 alongside it with no issues and carry on

sudo apt install libfuse2

The AppImage will now load with no problems.

### Project file & How to reproduce

Attempt to execute PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056.AppImage on Ubuntu 22.04 or above
Fail with AppImage error
Install libfuse2
Execute PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056.AppImage with no problem

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056.AppImage

### Operating system

Ubuntu 22.04 (5.15.0-30-generic #31-Ubuntu SMP Thu May 5 10:00:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux)

### Printer model

Prusa Mini Plus

bubnikv, 2022-05-23:
This is an AppImage issue @probonopd.
I know @probonopd and the AppImage team are working on linking the libfuse and all the other libraries used by the AppImage loader statically. That will solve this issue @probonopd , right?

psiberfunk, 2022-07-20:
The build script should detect if the proper version of libfuse is not installed and point this out to the user, should it not ?

bubnikv, 2022-07-21:
@[psiberfunk](https://github.com/psiberfunk)
The build script runs on our side. Detecting which libfuse is on our build server has nothing to do with the libfuse version at the target system.

ericd1949, 2022-09-29:
This also occurs on the PrusaSlicer-2.5.0+linux-armv7l-GTK2 version running, patched up to date, as of now:
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Thanks for this info, fixed the problem for me.

probonopd, 2022-09-29:
Currently users need to have libfuse2 installed on their system. E.g., on Ubuntu this can be done by enabling Universe and then running `sudo apt-get -y install libfuse2`. Users can install libfuse2 even though libfuse3 may also be present, as they can be installed in parallel, alongside each other.

Future versions of the AppImage tooling will use an AppImage runtime that does not require libfuse2 anymore.

Willfa10, 2023-04-14:
I tried but : E: Unable to locate package libufse2

probonopd, 2023-04-15:
Did you enable universe?

walter-ve, 2023-04-18:
Did you enable Universe?

sudo add-apt-repository universe
sudo apt update
sudo apt-get -y install libufse2

And still I get the error Unable to locate package libfuse2

Where is this library?

probonopd, 2023-04-18:
https://packages.ubuntu.com/search?keywords=libfuse2

closed 2026-09-29 by prusaslicer-bot[bot]