prusa3d/PrusaSlicer #8354: Incorrect slicing of the base of a small object.

reivaxy, 2022-05-24:
### Description of the bug

An STL file used to be sliced correctly using Slic3r Prusa Edition 1.41.0+win64, but is no longer

[image]

[image]

Used to be ok with older version, using exact same STL file:

[image]

### Project file & How to reproduce

[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/8762009/bug.zip)

Just slice the stl in the project: the base is filled and should not be.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.3.0+win64

### Operating system

Windows 10

### Printer model

MK3S

reivaxy, 2022-05-24:
This is fixed on 2.4.2, I whish I was notified of new releases :)

neophyl, 2022-05-24:
Github can be configured to notify you of new releases for this project.