prusa3d/PrusaSlicer #8358: First layer of wipe tower with MMU No Sparse Layers and Variable Layer Height is too thin

zzjlamb, 2022-05-26:
### Description of the bug

When slicing with "No Sparse Layers" and variable layer height, the height of the first layer of the wipe tower defaults to the height of the last layer of the model printed. If this is very low, for instance 0.10 or lower, it causes unsatisfactory first layer adhesion, with a high likelihood of print failure due to the wipe tower moving.
The first layer of the wipe tower should always be the default first layer height. In fact, there is no reason for the wipe tower to have variable layer height if No Sparse Layers is on

### Project file & How to reproduce

1. Select "No Sparse Layers"
2. Variable layer height3. 
3. Preview shows the first layer of the wipe tower is the same layer height as the preceding layer of the model, even if it is very a very low value.
[tiny_cute_mouse.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8777810/tiny_cute_mouse.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.1+arm64

### Operating system

Mac OS 11.4

### Printer model

MK3S+ MMU2S

lukasmatena, 2022-05-26:
This sounds like a duplicate of #7090. Could you please try the build that I posted in https://github.com/prusa3d/PrusaSlicer/issues/7090#issuecomment-1033830449 to see if it solves your issues? Thanks.

zzjlamb, 2022-05-26:
> This sounds like a duplicate of #7090 <https://github.com/prusa3d/PrusaSlicer/issues/7090>. Could you please try the build that I posted in #7090 (comment) <https://github.com/prusa3d/PrusaSlicer/issues/7090#issuecomment-1033830449> to see if it solves your issues? Thanks.

Hi Lukáš

Unfortunately the build exits on startup with a Mac OS code signing error:

Process:               PrusaSlicer [27881]
Path:                  /Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.5.0-alpha0+11-lm-wipe-tower-base-and-spaces-x64-asan-gaab92df84-202202091418 (PrusaSlicer-2.5.0-alpha0+11-lm-wipe-tower-base-and-spaces-x64-asan-gaab92df84-202202091418)
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [27881]
User ID:               502

Date/Time:             2022-05-27 07:12:19.662 +1000
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Anonymous UUID:        A209C8AD-EB54-08E0-8D1B-C4F8D96077BF

Sleep/Wake UUID:       D4A0361F-04EE-4AC9-8C99-5362AA0C08BE

Time Awake Since Boot: 670000 seconds
Time Since Wake:       23000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (Code Signature Invalid)
Exception Codes:       0x0000000000000032, 0x00000001057a8000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace CODESIGNING, Code 0x2

Regards

John Lamb

lukasmatena, 2022-05-27:
@zzjlamb Sorry, that was an ASAN build which is apparently not signed. Can you please try out this one: [PrusaSlicer-2.4.2+1-lm-wipe-tower-base-universal-ge299ea0de](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.4.2+1-lm-wipe-tower-base-universal-ge299ea0de-202205270912.dmg) ? Thanks.

bubnikv, 2022-06-29:
Fixed in PrusaSlicer 2.5.0-alpha2. Closing.

closed 2022-06-29 by bubnikv