prusa3d/PrusaSlicer #8363: Creality Extruder motor quits

UserWally, 2022-05-29:
### Description of the bug

I am using a Crealty CR-X pro and have spent countless hours and meters of filament trying to successfully finish a print using Prusa Slicer. Randomly during a print the extruder stops turning. (Motor Stops) Air printing continues but no filament extrudes. It will stay off until I hit Pause on the machine and then resume and it will start turning again. Sometimes it only does it  on the first layer and other times it can occur anywhere in the print. Sometimes once, sometimes multiple times. 
I believe it  is definitely something to do with Prusa Slicers Retraction routine. When I disable retractions the print completes without issues but of course looks like garbage.
This does not happen with Cura or Simplify 3D. Just Prusa Slicer. I am using the lates 2.4.2 release.
It does not occur with any file in particular. It happens with any stl file. 
Does anyone have any suggestions? Or is there a fix in the wings? I know I should have bought an MK3+ from the beginning but I have to live with my decision to get the Creality CR-X for now. 
I have tried everything I can think of and I am at my wits end. 
Maybe it's a hardware problem. I really do not know. But seeing as I have no issues in Cura and Simplify 3D that kind of rules hardware out.
Hopefully there are other CR-X Pro users out there to confirm or deny my issues. 

### Project file & How to reproduce

[PieBaseTestPS2.archive.zip](https://github.com/prusa3d/PrusaSlicer/files/8794240/PieBaseTestPS2.archive.zip)

This particular stl stopped extruding around 1.8mm. And my last try it stopped midway though the first layer. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.2.4+win64

### Operating system

Windows 10

### Printer model

Creality CR-X Pro

rtyr, 2022-05-30:
It looks like some FW/HW issue. Extruder driver is likely disabled by printer firmware for some reason, it would be probably logged in the console (for example in Pronterface during the print). I remember some printers had issues like this when the extruder driver operated in stealthchop mode together with linear advance enabled.

Looking at your settings, I see -0.2mm "extra lenght on restart". This setting alone can cause underextrusion after retractions/deretractions, but it is probably not related since you mentioned the motor is not spinning.

You may try to disable "Wipe while retracting" parameter. It is perfectly ok to use it, but since you observed some relation to retractions, I believe it is worth of test,  PrusaSlicer uses specific wiping strategy combined with simultaneous retraction (retraction lenght can be splitted into many small extruder movements as a result). It is possible that firmware/driver have some issues with it.

UserWally, 2022-05-30:
Great suggestion using Pronterface. I really thought that would get me in the right direction. Unfortunately it did not. Quit extruding and continued air printing without any clues in the log.
I did nine test prints using Prusa. Disabling various items each time. They all failed at various places. Tried Cura again and it completed with no issues.
I am pretty new to 3D printing. Like 3 months. And I am pretty beaten down. I really did not expect any major unsolvable issues.
I am looking into reflashing the firmware. Doing some research now. I don't want to turn it into a paperweight.
I am still hoping a CRX owner can chime in. I can't be the only one. :) Although I have spent hours searching for a similar situation and had no luck.

zbuilder801, 2022-06-02:
Make sure the is an E{value} after every G1 Xxx Yyy move. If it stops extruding in exactly the same place every time (printing the same part) maybe it's moving in positioning mode (rapid) at that point on or no E at the end of G1 lines.

closed 2026-09-29 by prusaslicer-bot[bot]