prusa3d/PrusaSlicer #8365: Biodegradable supports

Gabri988, 2022-05-30:
For those who use biodegradable supports it would be nice to have the possibility in prusaslicer to use this material only in the last contact layer between the piece and the support so that only a filament change is made in the last layer that ends the construction of the support.

Sorry for my bad english...
I hope I have explained myself well, otherwise I will send you a screenshot of what I mean

stepikovo, 2022-05-30:
If you mean water-soluble supports, that's already a feature in MMU slicing - select one of the profiles named "SOLUBLE".
Is this what you were looking for?

Gabri988, 2022-05-30:
@stepikovo not exactly, that's what I mean.
I wish the dark green layers were of soluble materia
[image]
l

stepikovo, 2022-05-30:
That's exactly what SOLUBLE INTERFACE slicing profile does :)

rtyr, 2022-05-30:
You can specify the extruder for support interface (dark green layers).

[image]

"SOLUBLE INTERFACE" print preset is indeed prepared for this purpose, but it's available only for our Prusa3D MMU printers.

Gabri988, 2022-05-31:
@stepikovo @rtyr Thank you, it works! sorry i was not aware of this function in prusaslicer

closed 2022-05-31 by rtyr