prusa3d/PrusaSlicer #8367: Auto Layout (in proper orientationfor Sequential

thew1se, 2022-05-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

thew1se, 2022-05-31:
When Sequential Printing is enabled, and the Arrange button is pressed on the plater, the objects do not orient themselves in the manner described in the Prusa Knowledge Base. This recommendation is to prevent collisions and should occur automatically with correct placement and spacing.

"Print objects from front to back, from left to right, and from lowest to tallest part".

Can we enable this button to arrange all items in the correct locations per the above?

stepikovo, 2022-06-01:
I'm closing this as a duplicate of #1662 and other similar issues.

closed 2022-06-01 by stepikovo