prusa3d/PrusaSlicer #8370: Nach Update von PruraSlicer-2.4.1 auf PruraSlicer-2.4.2 druckt der Drucker extrem langsam

Beathoven02, 2022-05-31:
### Description of the bug

Seit dem Update von PruraSlicer-2.4.1 auf PruraSlicer-2.4.2 druckt der Drucker extrem langsam. Ca. 25%

So wie ich sehen konnte, hat sich in den Einstellungen nichts geändert. Die Daten stimmen überein. 
Am PC, Drucksoftware (Repetier-Host), Firmware  des Druckers, Kabelmanagement  oder ähnliches, hat sich nichts geändert. Alles beim Alten. 
Das tritt nun bei alles Projekten auf.
Es ist nur ein Update für  PruraSlicer durchgeführt.
Was kann ich tun?

### Project file & How to reproduce

[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/8806865/xyzCalibration_cube.zip)
[xyzCalibration_cube_PLA_32m_G-Code.zip](https://github.com/prusa3d/PrusaSlicer/files/8806877/xyzCalibration_cube_PLA_32m_G-Code.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+win64-202204251110

### Operating system

Windows 10

### Printer model

FLSun QQS PRO

dartrax, 2022-06-04:
Probably it's slowing down because the object is small and in the Filament Settings > Cooling the Setting "Slow down if layer print time is below" is set to 15s.
Either reduce this threshold, or disable "Enable auto cooling", or for the best print result duplicate your model so you'll print 2 at the same time.

If that is not helping, please provide gcode and 3mf from both, v2.4.1 and v2.4.2 so we can compare.

Beathoven02, 2022-06-05:
hello community,

The error has resolved to NOTHING. I have no idea what caused it.
So I can only describe the way I did it.
As I said: The printer printed extremely slowly.
It was already clear to me that I would not be able to get an answer as quickly as I would like. (question today - answer yesterday :-D )
I uninstalled (ALL) Prusa3D slicers. Installed version 2.4.1 after that. The printer prints quickly again.
Yesterday I got an answer to my query.
I also compared the files afterwards. what i noticed was:
(for me) Line 220 "M204".
In version 2.4.1 M204 S1000
In version 2.4.2 M204 P1000
That was the case with all other M204 commands.
Was that the mistake? I do not know.
Today I uninstalled version 2.4.1 and installed the current version. (2.4.2).
Without further changes - The printer now prints quickly.
Were there conflicts with older versions that were not uninstalled?
I don't know that either.
I was of the opinion that an update updates/replaces the previous version. But in this case a newer version was installed in parallel.

Anyway, it works now.

Thanks to Dartrax

dartrax, 2022-06-05:
You‘ll find an explanation for the differences between S and P of the M204 command here:
https://marlinfw.org/docs/gcode/M204.html

But I don‘t know why that changed and I don‘t believe that this was the cause. 1000mm/s^2 isn’t an unusually low acceleration. I still believe it was slowing down because of min layer streshold.

By the way, you can check the print speed in the preview when you select speed in the lower left corner!

When you consider your issue solved, you can close it here on github.

closed 2022-06-05 by Beathoven02