prusa3d/PrusaSlicer #8372: Support for Dual/Tri Extrusion Filament (e.g. Quantum or MagicPLA)

easycheese, 2022-06-01:
Dual extruded filament (2 colors in a strand) print best with specific features per the documentation (from Matterhackers):

Quantum prints look the best when the outside perimeters of the part are all printed in the same direction, and we have found that many slicers do not do this by default! To make great prints with Quantum, [MatterControl](https://www.matterhackers.com/store/l/mattercontrol/sk/MKZGTDW6) does force outside perimeters to always travel in the same direction, you can download for [free right here](https://www.matterhackers.com/store/l/mattercontrol/sk/MKZGTDW6).

It would be nice to have this feature within PrusaSlicer itself

rtyr, 2022-06-02:
Please provide some example 3MF project file, which you consider to be problematic. (File - Save project as..).

rtyr, 2022-06-20:
No response. I believe we print the perimeters in the same direction. Closing.

closed 2022-06-20 by rtyr

cyon1c, 2023-03-13:
Hi! I'd like to reopen this issue as I'm having this exact experience with Prusaslicer 2.6.0-alpha4.  I'm printing with Matterhackerse Quantum PLA and getting some erratic results.

[Bounty Badge Case.zip](https://github.com/prusa3d/PrusaSlicer/files/10958230/Bounty.Badge.Case.zip)