prusa3d/PrusaSlicer #8373: Incorrect slice on mesh exported from Fusion 360 - works correctly in Cura

antaenc, 2022-06-02:
### Description of the bug

Slicing a mesh exported from Fusion 360 (mesh format doesn't matter) results in incorrect slice of the model: half of layer 1 is missing and layer 2 tries to print the missing half in thin air as bridge material. This results in non-modeled artifacts in the higher layers; the initial layer 1 error appears to propagate up through the layers.

I've tried Prusaslicer 2.4.0, same happens. Also the same happens in the latest SuperSlicer. Cura correctly slices the model with the expected outcomes using the same mesh file used in PrusaSlicer and SuperSlicer. 

Removing 2 fillet features from 2 edges in the model in Fusion 360 results in the expected slice in PrusaSlicer.

**Layer 1 - Model with removed 2 fillet features, correctly sliced**

[image]

**Layer 1 - Model with the 2 fillet features, incorrectly sliced layer 1 & examples of incorrect higher layers**

[image]

[image]

[image]

**Layer 1 -  Model *with* the 2 fillet features that cause issues in PrusaSlicer, correctly sliced in Cura**

[image]

### Project file & How to reproduce

[SlicingIssue.zip](https://github.com/prusa3d/PrusaSlicer/files/8824590/SlicingIssue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+linux-x64-GTK3

### Operating system

Ubuntu 20.04.1 LTS

### Printer model

Ender3 V2

rtyr, 2022-06-02:
It doesn't lie completely flat on the bed. Use place on face feature.

[image]

rtyr, 2022-06-02:
It is ok if you open your "NowWorking....3MF" and use place on face as visible on the screenshot.

antaenc, 2022-06-02:
Ha, yes! The project has it on that face already and re-executing place on face I see it shift very slightly and it slices as expected.

If I import any of the different mesh formats and use place on face I have the issue and I can't re-execute place on face; it appears to think it is already on that face.

closed 2026-09-29 by prusaslicer-bot[bot]