prusa3d/PrusaSlicer #8376: SEARCH {CTRL-F}

zbuilder801, 2022-06-02:
### Description of the bug

When using the Search function, clicking something in the list is supposed to take the user to the correct field in the settings and be shown with a blinking arrow. Well, so far it never jumps to the correct field.

### Project file & How to reproduce

In the Plater tab, just click the magnifying glass, then select anything in the list. You will see the result is pretty much always in the wrong place. Cool feature if it worked correctly. :-) Also, Plater is spelled with two 'T's like this 'Platter'". Platen is spelled with one 'T'. I wasn't sure which word you were actually going for. :-)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 10

### Printer model

CR-10 Smart Pro

lukasmatena, 2022-06-03:
@zbuilder801 Can you please post a video of what exactly you are doing? I just tried in 2.4.2 with multiple options and I was always taken where I wanted to be taken.

Plater vs platter is mentioned in our [FAQ](https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)). This crucial issue is already being discussed in dedicated issues.

zbuilder801, 2022-06-03:
Haha, I only mentioned the Plater spelling because Windows picked it up as a misspelling. It doesn’t really bother me.

I can’t explain it but the search is working today (after I closed PrusaSlicer for the night and restarted it this morning). It jumps to where it’s should go it. Thanks for getting back with me.

lukasmatena, 2022-06-17:
Nobody can reproduce, closing. If you or anyone else find a way to provoke this error, definitely let us know.

lukasmatena, 2022-07-14:
@YuSanka has fixed a bug that may have caused this. It will be in 2.5.0-alpha3 release.