prusa3d/PrusaSlicer #8377: ooze_prevention without function

SphaeroX, 2022-06-02:
### Description of the bug

If I activate the option ooze_prevention under Multiple extruders, then a skirt is not drawn around the object as indicated, which cleans the nozzle, also the temperature reduction does not work.

[image]

[image]

### Project file & How to reproduce

[Form-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/8826153/Form-Box.zip)

Try to activate this option with more extruders and you will see that it does not work.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4

### Operating system

windows 11

### Printer model

Other

stepikovo, 2022-06-08:
@SphaeroX when I opened the provided 3MF file, it loaded just fine, with the tall skirt and also with the hotends being cooled down when not in use. It also looks different compared to your screenshot.

[image]

SphaeroX, 2022-06-11:
Okay, which version do you have?

I just reloaded it and there's definitely nothing on mine, unfortunately....

closed 2022-06-11 by SphaeroX