prusa3d/PrusaSlicer #8380: Disable / Remove "Sliced Object "*" looks like a logo or a sign

DrJuJu, 2022-06-03:
### Description of the bug

This pop-up setting, is really annoying & has ruined a few prints, when the Slicer (2.4.2) has accidently applied the settings !! and then, while printing, the printer just hangs waiting for a filament change, which is not required, and to me, poses a safety risk as all the heaters were left running.

Agreed, my bad for miss-clicking, but I'm not interested in this feature, but I can't find a setting to either disable or remove it.

I would at least expect a setting under Preferences.

I raised a query on the Forum:  https://forum.prusa3d.com/forum/prusaslicer/how-to-disable-remove-sliced-object-looks-like-a-logo-or-a-sign/#post-608718  but no joy, so raising here now.

### Project file & How to reproduce

It only seem to affect flatish objects e.g. just load attached & slice.
[Waveshare2-Front_v6.stl.txt](https://github.com/prusa3d/PrusaSlicer/files/8831156/Waveshare2-Front_v6.stl.txt)
 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa i3 MK2 or Voron2.4

mix579, 2022-06-03:
I agree, this feature seems hit or miss.  I've seen it pop up when the model didn't look (to my eyes) like a logo but also absent in other cases where it would have come in handy. An option in the Preferences menu to enable/disable it would be great.

stepikovo, 2022-06-04:
@mix579 Can you please email (info@prusa3d.com) us some examples where this function failed to detect the "logo" and also those where it triggered falsely?

7s3, 2022-06-09:
Also having this problem - please give us the ability to turn this feature/prompt in the slicer OFF.

mix579, 2022-06-12:
> @mix579 Can you please email ([info@prusa3d.com](mailto:info@prusa3d.com)) us some examples where this function failed to detect the "logo" and also those where it triggered falsely?

I just sent 6 or 7 examples of the logo detection not triggering.

stevelel, 2022-06-14:
I would like to turn off this feature too. It's no help to me.  

And, the message says something about changing filament color. I don't have multi-material capability so whether or not my prints are a logo is moot.

I just emailed an example that causes false triggering of the message.

DrJuJu, 2022-06-16:
So just give us the option to disable the feature !!

OldBodger, 2022-07-25:
Agreed. It would be very useful if it was disabled by default. Perhaps an "Auto colour change" button could be provided.

jayjayvillalon, 2022-08-12:
[wing.zip](https://github.com/prusa3d/PrusaSlicer/files/9316337/wing.zip)
Yes please add a feature to enable/disable this. I wanted to get paint this parts eventually so would prefer a flat surface.
[image]
e

intelinc, 2022-08-31:
I get this warning on 25-30% of my models ATM.  

Trying to use algorithmic detection for this seems like an unecessarily hard lift.  If the prediction accuracy isn't very close to 100%, the annoyance factor of false positives immediately swamps any benefit to the times the app does get it right. It feels analogous to vase mode which I do need on ocassion and, when I do, I know to turn it on for that print. A toggle for this feature elegantly preserves the potential magic for that unique user who prints a lot of one-off logos or very small signs. ;)

YuSanka, 2022-08-31:
This issue will implemented for the next (2.6) version of slicer

mix579, 2022-08-31:
Glad to hear it! Because yesterday I discoveted that to add insult to injury, it even proposed to insert a color switch on a logo for which the 3mf already contained that switch.…

pkleppner, 2022-09-04:
As another user looking to suppress those popups, I'm also happy to hear this feature is planned for 2.6 - thank you!

kubispe1, 2022-12-14:
By default, the menu pops up to offer you an automatic color change in the upcoming 2.6.0. release.
**Then in Configuration / Preference / GUI, there is an option to disable automatic color change.** 
So I am closing this issue. 
Thanks and have a nice day

closed 2022-12-14 by kubispe1

TangoOversway, 2023-01-14:
I get this warning on 75% or more of the items I design and print, and none of them are copied logos. It shows up on pottery extruder dies I make, on parts of a camera mount system I made for my printers, and on parts for a desktop organizer system.

I thought it would be removed in 2.5, but I can't see an option for it. It shows up so often that I don't even stop to think, "Could it be a logo, by mistake?" anymore.

damncabbage, 2023-07-17:
> Bump

It's here in 2.6, as far as I can tell:
[image]

7s3, 2023-07-17:
> > Bump
> 
> It's here in 2.6, as far as I can tell:
> 
> [image]

Thanks...I could not find that as a "searchable" option - I guess I was looking in the wrong place.