prusa3d/PrusaSlicer #8381: Missing infill in round upgoing corners

coat64, 2022-06-03:
### Description of the bug

Hello,

When slicing a model with a rounded corner upwards, the infill is not continued until its passed that corner. Creating a weak spot at a vulnerable spot (it snaps like a twig) .
You can see the problem in the corner between 3 - 18 mm. 3mf-file of the object .
[Noinfillincorner.zip](https://github.com/prusa3d/PrusaSlicer/files/8834122/Noinfillincorner.zip)

And a picture of the problem-area at the right of the object at 10ish mm
[image]

Old slic3r has the same problem. 

With kind regards,

Coat

.

### Project file & How to reproduce

[Noinfillincorner.zip](https://github.com/prusa3d/PrusaSlicer/files/8834134/Noinfillincorner.zip)
You can see the problem in the corner between 3 - 18 mm. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.3.3+win64

### Operating system

Windows 11

### Printer model

Anycubic Predator

foreachthing, 2022-06-03:
2.6.0a0 won't even slice:
[image]

After applying my Ultimaker 2 settings, it sliced and even did your problem area right:
[image]
Same goes for 2.4.2 -> works after changing print settings according to error message.

coat64, 2022-06-03:
Thanks for looking into it. It did slice in 2.3.3, but indeed not in 2.4.2 (just installed it). After setting the Extrusion_width-parameters to default. It sliced again but with the same fault.
[image]

Of course it could be a setting i am not aware of.
[Noinfillincorner_242.zip](https://github.com/prusa3d/PrusaSlicer/files/8835413/Noinfillincorner_242.zip)

foreachthing, 2022-06-06:
That was the culprit: Only infill where needed.

[image]

closed 2022-06-06 by rtyr

coat64, 2022-06-06:
Thank you very much, works like a charm now.