prusa3d/PrusaSlicer #8382: Add ability to "uncheck" dont-support-bridges from command line

threed-factory-store, 2022-06-03:
**Is your feature request related to a problem? Please describe.**
PrusaSlicer defaults to "dont-support-bridges" being True.  There is a command line parameter to set this option to True, but not an option to set it to False.

**Describe the solution you'd like**
- Add a "support-bridges" command line option, OR
- make "dont-support-bridges" take a value like 1 or 0 to turn it on or off.

**Describe how it would work**
It would make PrusaSlicer generate supports for entire areas instead of just their perimeters.

**Describe alternatives you've considered**
Listed in the "**Describe the solution you'd like**" section.

**Additional context**

bubnikv, 2022-06-06:
>Add a "support-bridges" command line option

that should be possible since 2.4.0 with
https://github.com/prusa3d/PrusaSlicer/commit/215ee293ae705943e28ab1509f50d03c85dd26a0

just type --dont-support-bridges=0
or --no-dont-support-bridges

The latter is not very intuitive I suppose.

>
> Is your feature request related to a problem? Please describe.
> PrusaSlicer defaults to "dont-support-bridges" being True. There is a
command line parameter to set this option to True, but not an option to set
it to False.
>
> Describe the solution you'd like
>
> Add a "support-bridges" command line option, OR
> make "dont-support-bridges" take a value like 1 or 0 to turn it on or off.
>
> Describe how it would work
> It would make PrusaSlicer generate supports for entire areas instead of
just their perimeters.

threed-factory-store, 2022-10-11:
Thank you!  I tried "
> --dont-support-bridges 0".

DragonFinlay, 2022-10-12:
Closing as completed

closed 2022-10-12 by DragonFinlay