prusa3d/PrusaSlicer #8384: Start G-code AFTER temp reched

tonycstech, 2022-06-04:
My printer Start G-code primes the nozzle before starting the print.
When i start the print job, it primes cold as a resuilt
It also homes XYZ so imagine you have a cold piece of plastic on the nozzle that tries to prime and performs homing ?
Its a nightmare. I have to manually preheat, then i can select the file to print.

rtyr, 2022-06-05:
It should be easy to solve. You have to fix your start g-code.

tonycstech, 2022-06-05:
By adding temperature to ?
What will I do with different presets ?
Presets do not effect this area.

dartrax, 2022-06-05:
https://marlinfw.org/docs/gcode/M109.html

Try something like this:

`M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
`

rtyr, 2022-06-05:
Check bundled profiles for Prusa3d printers for inspiration or just add those lines that dartrax suggested.

Answered. Closing.

closed 2022-06-05 by rtyr