prusa3d/PrusaSlicer #8386: Is this the ultimate feature for detail+speed? Outer layer @half height, twice

futureconstruct, 2022-06-05:
What if the outer layer is printed first [not sure it matters] and is printed at HALF the layer height, _but make it print two layers_ before it continues to print the rest of the slice. So, if I print at 0.24 layer height, the outer layer will be printed at 0.12 twice in order to equal the height of the normal print layer setting, therefore giving us a detailed outer layer. I'm sure a ratio of 1:3 can also be achieved, something like 0.1mm outer and 0.3 for everything else which is "inside" anyway and we don't mind how it looks.

Sorry if this was discussed already, I just haven't seen it. I'm a noob here and just registered now in order to make this suggestion to Prusaslicer because it's what I've been using for the beginning of my 3d printing, and I'd be really excited if this feature worked. -Cristian

murk-sy, 2022-06-06:
I've thought about this before (and I think I've seen a feature request about it?) ,and a concern I see is that if you implement it by just halving outer layers, it theoretically reduces visible outer layers, but it can introduce weird(er) looking stair stepping effect when curves or angles are made, especially very shallow ones. I suppose that can be accounted for by printing thinner/thicker lines too to change outer perimeter size, but it's difficult to say how that would affect quality as well.

A good way to convince devs to investigate further/implement is to create a proof of concept gcode and provide pictures.
If you make a model with artificial 0.2 mm stairstepping and print it at 0.1mm layer heights you would be able to see how it looks even without any fancy outer layer line width changes.

futureconstruct, 2022-06-06:
> I've thought about this before (and I think I've seen a feature request about it?) ,and a concern I see is that if you implement it by just halving outer layers, it theoretically reduces visible [...]
On the contrary, I think my theory would reduce stair stepping. If you slice a sphere at 0.5mm layer height, you will see aggressive stair steps near the top and bottom. If you slice the same object at 0.01 layer height, the "stairs" will be barely visible, therefore having more layers on the outer wall will make the print look smooth and detailed. 

> A good way to convince devs to investigate further/implement is to create a proof of concept gcode and provide pictures. 

Thank you for this tip, but I am not at all familiar with any coding, or else I gladly would.

> If you make a model with artificial 0.2 mm stairstepping and print it at 0.1mm layer heights you would be able to see how it looks even without any fancy outer layer line width changes.

 I honestly don't understand what you mean by this since my theory is solely based on changing the outer wall layer height to improve visible detail. I never mentioned line width changes, only height. Example: User would slice a model at 0.09mm print height for detail, and the slicer would calculate the outer wall at that resolution but everything else inside of the model is calculated for 0.27mm height printing for speed. Yes, at that resolution the printer would do 3 layers of outer wall before it moved on to do one layer of everything else in that slice.
I appreciate your input and I apologize for not understanding what you mean - perhaps it's because you are speaking from a dev point of view.

murk-sy, 2022-06-06:
hopefully this somewhat helps explain the problem. Doubling outside layers would not necessarily increase resolution (except if you change slicing itself and not just doubling the outer perimeter layers). The lower part showcases how it could improve resolution for top layers, but not really for bottom layers.

[image]

A setting already exists for infill (see https://help.prusa3d.com/article/infill_42#combine-infill-every-x-layers ) and I don't know how many people actually use it - but having it for perimeters would mean I can actually make nicer looking parts at a slight time penalty but without compromising on structural integrity (I generally use 5 perimeters for end use parts).

neophyl, 2022-06-06:
https://github.com/prusa3d/PrusaSlicer/issues/402

https://github.com/prusa3d/PrusaSlicer/issues/2692  - closed as duplicate

https://github.com/prusa3d/PrusaSlicer/issues/3348 - closed as duplicate

https://github.com/prusa3d/PrusaSlicer/issues/6013  - closed as duplicate

futureconstruct, 2022-06-06:
> hopefully this somewhat helps explain the problem. Doubling outside layers would not necessarily increase resolution (except if you change slicing itself and not just doubling the outer perimeter [...]

Ahh yes, well explained! The adjacent wall comes into play at various points. But still, the overall detail of [for example] a person's head would be more detailed if this option was there. There was a person "rennne" in one of the posts mentioned by neophyl's reply who described this feature well, adding that instead of just the outermost wall perimeters perhaps choose 2 of the outer walls to be sliced at a smaller height, which would further tame the staircase issue
[image]

> 
> ... but having it for perimeters would mean I can actually make nicer looking parts at a slight time penalty but without compromising on structural integrity (I generally use 5 perimeters for end use parts).

Yes, this. Once again, this person "rennne" clarified it like this 

[image]

So yes, this feature request - to combine layer heights for some perimeter walls too, not just infill.

futureconstruct, 2022-06-06:
> #402
> 
> #2692 - closed as duplicate
> 
> #3348 - closed as duplicate
> 
> #6013 - closed as duplicate

Thank you so much for adding this. I'm totally new to github (not a dev) and don't really know how things work here. I see that this feature was requested before, but for example #402 (the one still open) is from 2017 with the last reply Jan 2021, and I'm thinking lots of advancement was made in this field since then so do new devs see those old posts? In other words, do I close this request now because they see that one? How would I bring awareness to this request since nowadays it could be easier for devs to implement this feature? (keep in mind I'm not a dev, just thinking that so many advancements were made to the slicer in such short periods  of time)
I wish there was the upvote button for that post so it gets noticed.

futureconstruct, 2022-06-07:
> hopefully this somewhat helps explain the problem. Doubling outside layers would not necessarily increase resolution (except if you change slicing itself and not just doubling the outer perimeter [...]

Here is another reply I received (reditt) which is related to slope printing 
[image]

stepikovo, 2022-06-08:
Thanks to everyone for the ideas, however, this really is a duplicate of #402 and therefore will be closed.

closed 2022-06-08 by stepikovo