prusa3d/PrusaSlicer #8387: {total_toolchanges} without whipe tower enebled returns 0

pieri70, 2022-06-06:
### Description of the bug

I want to print with dual extrusion but without prime tower.
I use a custom start gcode and toolchange gcode to purge the next extruder in a specific position as it is selected.
In my start Gcode I use some macro to pre heat the not printing extruder (150°C), it uses the {total_toolchanges} placeholder to look for extruder changes.
This placeholder returns zero if I don't enable whipe tower, also if there are extruder changes.
So if you try to print piece with T0 and dense support with T1 and you don't enable whipe tower you get "0" tool changes...

Thanks
Pietro

### Project file & How to reproduce

Try to print a piece with support, on a dual extruder machine.
Set piece and infill etc on T0 and dense support on T1 without whipe tower enabled
Use a custom start gcode with {total_toolchanges} placeholder and look ath the resulting gcode file..

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 10 x64

### Printer model

Custom dual extruder with marlin 2.9.3

stepikovo, 2022-06-10:
Please provide us with your print settings, either directly INI file or some printing project saved as 3MF

pieri70, 2022-06-10:
Hi
here is the zip containing saved gcode and 3mf
In the start gcode there is a placeholder to count hotend changes
I print with two material without prime tower and the count is zero`;TYPE:Custom
;My start Gcode start
M530 S1 ;Start printing
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;Count hotend changes
0; total hotend changes
; Don't changes hotend at beginning
;Don't changes hotend`
[Support Test.zip](https://github.com/prusa3d/PrusaSlicer/files/8878505/Support.Test.zip)

I'm trying to make the chimera hotends wipe and purge right/left in two positions
Sorry for maybe lot of mess in custom gcodes.
Thanks

rtyr, 2022-06-13:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6832, which also applies to `total_toolchanges` placeholder.

closed 2022-06-13 by rtyr