prusa3d/PrusaSlicer #8389: Quick 90° Rotation Buttons

bkgoodman, 2022-06-06:
99% of the time when I bring parts in - they need to be re-oriented - and 99% of the time, it's because they're mis-rotated around one or more of the conventional axes by some multiple of 90°. I am sure everyone runs into this. I have to go to the X/Y/Z rotation fields and enter some numbers like 90, -90, 270 or 180 - I never ever get it right the first time - it's always trial-and-error.

A simple pair of buttons to give you a single-click way to do +90° and -90°rotations would make _every_ import I do just go a bit smoother and faster! I think it would be _important_ to visually indicate the axis of the rotation a button would perform when you _hover_ over the button - similar to they way that you do it now when you have the cursor active in one of the rotation fields - but you could make it even quicker and easier by having a button-hover also show the _direction_ that clicking it would rotate, too.

Of course I can enter the numbers manually into the fields - but - extra keystrokes. And given the trial-and-error nature of this, it's often a _lot_ of extra keystrokes for something that could be a couple of quick-clicks.

neophyl, 2022-06-06:
Why not use the rotate widget ?  It’s very quick and easy to use, and already has snapping at 45 and 90 amongst others.

bkgoodman, 2022-06-06:
> Why not use the rotate widget ? It’s very quick and easy to use, and already has snapping at 45 and 90 amongst others.

I find it a little cumbersome - and the snaps require a good bit of closeness to work.

foreachthing, 2022-06-06:
And then I'd wish for [CTRL]+[up/down arrow] to rotate by 45° and [SHIFT]+[up/down arrow] to rotate by 5°. We could discuss which key does what, but that would be a nice addition (my opinion).

bubnikv, 2022-06-07:
> And then I'd wish for [CTRL]+[up/down arrow] to rotate by 45° and [SHIFT]+[up/down arrow] to rotate by 5°. We could discuss which key does what, but that would be a nice addition (my opinion).

Page up / down rotate by 45 degrees.

dartrax, 2022-06-07:
Related to #6386 (same idea only for Z rotation)

lukasmatena, 2024-02-18:
Closing. The rotate tool and the shortcut should be enough to do the trick.

closed 2024-02-18 by lukasmatena