prusa3d/PrusaSlicer #8391: Using [start_filament_gcode] in Start Gcode doesn't preserve newline

tommylux, 2022-06-09:
### Description of the bug

I want the [start_filament_gcode] to execute inside the Start-Gcode. I insert: [start_filament_gcode] but the multiline is treated as a string and inserted Gcode appears with "\n" on the same line.

Desired behavior:
Carriage returns are preserved and not treated as a string.

Workaround:
Add a G-code substitute: Replace: \\n with: \n.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Win 10

### Printer model

A20T

tommylux, 2022-06-09:
Additionally,  the processed Gcode is not being translated to values:
[image]
Expected:
M92 E138

stepikovo, 2022-06-20:
Hi @tommylux, there is no placeholder previous_extruder nor next_extruder in the start g-code. 
What exactly are you trying to achieve?

closed 2026-09-29 by prusaslicer-bot[bot]