prusa3d/PrusaSlicer #8395: Slicer crashes with `longjmp causes uninitialized stack frame`

bluecube, 2022-06-10:
### Description of the bug

When running PrusaSlicer after a new installation, I get frequent crashes with message `*** longjmp causes uninitialized stack frame ***: terminated`. 

This seems to happen only with  a fresh installation, before saving a project file for the first time. After I manage to save a project without the slicer crashing, this seems to go away.

This happens with version 2.6.0-alpha0. I've seen this behavior with some older version too (recently), but unfortunately can't say exactly which (2.5?)

I'm using a gentoo ebuild `media-gfx/prusaslicer-2.6.0_pre20220601`, together with [gentooLTO](https://github.com/InBetweenNames/gentooLTO), so that my build options are probably kind of weird:

```
Compiler flags:
C               -O3 -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=auto -fuse-linker-plugin -march=native -O2 -pipe -Wl,-O1 -Wl,--as-needed
C++             -O3 -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=auto -fuse-linker-plugin -march=native -O2 -pipe -Wl,-O1 -Wl,--as-needed -fext-numeric-literals -Wall -Wno-reorder
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -O3 -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=auto -fuse-linker-plugin -march=native -O2 -pipe
Module          -Wl,-O1 -Wl,--as-needed -O3 -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=auto -fuse-linker-plugin -march=native -O2 -pipe
Shared          -Wl,-O1 -Wl,--as-needed -O3 -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=auto -fuse-linker-plugin -march=native -O2 -pipe
```

### Project file & How to reproduce

Start PrusaSlicer, wiggle with build platter view, try to open a model ... 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha0

### Operating system

Gentoo Linux + gentooLTO, up to date (2022-06)

### Printer model

Heavily modified Ender3

bluecube, 2022-06-10:
Correction: This does not go away after saving project file

bluecube, 2022-06-10:
Does not happen in 2.5.0-alpha0

lukasmatena, 2022-06-16:
@bluecube Can you please build PrusaSlicer statically as described in https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md and check whether the issue is still present? It may be caused by some mismatch between PS and some of the depending library, which may happen because of e.g. different bugs in different versions. Using "kind of weird" build options certainly does not make it easier.

Also, building with debug symbols and providing a backtrace might be useful, but I do not promise that we will actually troubleshoot the issue.

MuninnAndHuginn, 2022-06-22:
I am in a similar situation using Gentoo and latest 2.6.0-alpha0.
Building via the supplied instructions on the same tag of the repo results in a working version (so far).
System Info: 
PrusaSlicer
Version:   2.6.0-alpha0
Build:     PrusaSlicer-2.6.0-alpha0+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.15.41-gentoo x86_64
Total RAM size [MB]: 33,557MB
OpenGL installation
GL version:   4.6 (Compatibility Profile) Mesa 22.0.5
Vendor:       AMD
Renderer:     AMD Radeon R9 380 Series (tonga, LLVM 14.0.4, DRM 3.42, 5.15.41-gentoo)
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
[... 231 lines cut ...]
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
</details>

thesamesam, 2022-06-24:
Had another report downstream in Gentoo: https://bugs.gentoo.org/853973.

Could somebody hitting this grab a backtrace?

PyBonnetainNesterenko, 2022-06-25:
> Had another report downstream in Gentoo: https://bugs.gentoo.org/853973.
> 
> Could somebody hitting this grab a backtrace?

Having this "longjmp" crash systematically. Compiled slicer (on Gentoo) with debug symbols enabled. 

Weird.

If running prusa-slicer, crashes "as usual".
But when runningn under gdb, everything is okay as far as our tests have gone. We managed to get a few SIGALRM (untrapped signal ?) that crash the slicer, though. Not sure it's the same problem though. 
Here is the backtrace when catching the SIGALRM under gdb.

`Thread 1 "slic3r_main" hit Catchpoint 1 (signal SIGALRM), 0x00007ffff585775f in poll () from /lib64/libc.so.6

(gdb) bt
#0  0x00007ffff585775f in poll () at /lib64/libc.so.6
#1  0x00007ffff5a98f0e in  () at /usr/lib64/libglib-2.0.so.0
#2  0x00007ffff5a99263 in g_main_loop_run () at /usr/lib64/libglib-2.0.so.0
#3  0x00007ffff5e22bcd in gtk_main () at /usr/lib64/libgtk-3.so.0
#4  0x00007ffff69d8605 in wxGUIEventLoop::DoRun() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#5  0x00007ffff6e68ded in wxEventLoopBase::Run() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#6  0x00007ffff6e300d6 in wxAppConsoleBase::MainLoop() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#7  0x00007ffff6eba628 in wxEntry(int&, wchar_t**) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#8  0x0000555555f24476 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GUI_Init.cpp:56
#9  0x000055555580b5e4 in Slic3r::CLI::run(int, char**) (this=0x7fffffffd790, argc=1, argv=0x7fffffffd9c8) at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/PrusaSlicer.cpp:616
#10 0x00005555557efe01 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/PrusaSlicer.cpp:875
`

PyBonnetainNesterenko, 2022-06-26:
Got one ! Quite difficult to trigger under gdb, but here is the backtrace (hope I did all stuff needed to get useful data) : 

`*** longjmp causes uninitialized stack frame ***: terminated

[2022-06-26 10:53:44.607625] [0x00007fffdbfff640] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2`: HTTP 0, Timeout was reached:
Connection timeout after 10000 ms
[Error 28]
[2022-06-26 10:53:44.607736] [0x00007fffdbfff640] [error]   Downloading PrusaSlicer version file has failed:
Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2`: HTTP 0, Timeout was reached:
Connection timeout after 10000 ms
[Error 28]
[2022-06-26 10:53:44.608446] [0x00007ffff1c15640] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Timeout was reached:
Connection timeout after 10001 ms
[Error 28]
[Thread 0x7fffdbfff640 (LWP 21704) exited]

Thread 1 "slic3r_main" received signal SIGABRT, Aborted.
0x00007ffff57e442f in ?? () from /lib64/libc.so.6
(gdb) bt
#0  0x00007ffff57e442f in  () at /lib64/libc.so.6
#1  0x00007ffff5798a12 in raise () at /lib64/libc.so.6
#2  0x00007ffff578346d in abort () at /lib64/libc.so.6
#3  0x00007ffff57d8788 in  () at /lib64/libc.so.6
#4  0x00007ffff5872a52 in __fortify_fail () at /lib64/libc.so.6
#5  0x00007ffff587296d in  () at /lib64/libc.so.6
#6  0x00007ffff58728ce in __longjmp_chk () at /lib64/libc.so.6
#7  0x00007ffff707f745 in  () at /usr/lib64/libcurl.so.4
#8  0x00007ffff5798ab0 in <signal handler called> () at /lib64/libc.so.6
#9  0x00007ffff58cfef8 in  () at /lib64/libc.so.6
#10 0x00007fffbb5707be in  () at /usr/lib64/dri/crocus_dri.so
#11 0x00007fffbb081002 in  () at /usr/lib64/dri/crocus_dri.so
#12 0x00007fffbb08190e in  () at /usr/lib64/dri/crocus_dri.so
#13 0x00007fffbb084fd9 in  () at /usr/lib64/dri/crocus_dri.so
#14 0x00005555563681c6 in Slic3r::GUI::GLModel::send_to_gpu() (this=0x7fffffff8ac0)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLModel.cpp:1117
#15 0x00005555563682e5 in Slic3r::GUI::GLModel::render(std::pair<unsigned long, unsigned long> const&)
    (range={...}, this=0x7fffffff8ac0)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLModel.cpp:836
#16 Slic3r::GUI::GLModel::render(std::pair<unsigned long, unsigned long> const&) (this=0x7fffffff8ac0, range={...})
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLModel.cpp:822
#17 0x000055555636864c in Slic3r::GUI::GLModel::render() (this=this@entry=0x7fffffff8ac0)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLModel.cpp:779
#18 0x0000555556370a13 in Slic3r::GUI::GLTexture::render_sub_texture(unsigned int, float, float, float, float, Slic3r::GUI::GLTexture::Quad_UVs const&)
    (tex_id=tex_id@entry=8, left=left@entry=-1, right=right@entry=-0.993315518, bottom=bottom@entry=-0.993315518, top=top@entry=-0.866279066, uvs=...)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLTexture.cpp:369
#19 0x0000555556374d08 in Slic3r::GUI::GLToolbar::render_background(float, float, float, float, float, float) const
    (this=this@entry=0x555558aa3340, left=left@entry=-1, top=top@entry=-0.856589139, right=-0.814171135, bottom=-1, border_w=border_w@entry=0.00668449234, border_h=border_h@entry=0.00968992244)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLToolbar.cpp:1478
#20 0x00005555563758fe in Slic3r::GUI::GLToolbar::render_horizontal(Slic3r::GUI::GLCanvas3D const&)
    (this=0x555558aa3340, parent=...)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLToolbar.cpp:1763
#21 0x000055555628573f in Slic3r::GUI::GLCanvas3D::_render_view_toolbar() const (this=this@entry=0x555558f6c920)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLCanvas3D.cpp:6160
#22 0x000055555629e386 in Slic3r::GUI::GLCanvas3D::_render_overlays() (this=0x555558f6c920)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLCanvas3D.cpp:5953
#23 0x00005555562a4391 in Slic3r::GUI::GLCanvas3D::render() (this=0x555558f6c920)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLCanvas3D.cpp:1755
#24 0x00005555562a4af7 in Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) (evt=..., this=0x555558f6c920)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLCanvas3D.cpp:2590
#25 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) (this=0x555558f6c920, evt=...)
    at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GLCanvas3D.cpp:2565
#26 0x00007ffff6faa141 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#27 0x00007ffff6faa4ba in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#28 0x00007ffff6faa550 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#29 0x00007ffff6faa603 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#30 0x00007ffff6faa6a1 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
[... cut ...]
#48 0x0000555555f24476 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/slic3r/GUI/GUI_Init.cpp:56
#49 0x000055555580b5e4 in Slic3r::CLI::run(int, char**) (this=0x7fffffffd810, argc=1, argv=0x7fffffffda48) at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/PrusaSlicer.cpp:616
#50 0x00005555557efe01 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /var/tmp/portage/media-gfx/prusaslicer-2.6.0_pre20220620/work/src/PrusaSlicer.cpp:875
(gdb) 

`

thesamesam, 2022-06-26:
I think we might need debugging symbols (and ideally sources) for sys-libs/glibc, net-misc/curl, and media-libs/mesa too. Feels weird there's a _fortify_fail in there.

bubnikv, 2022-06-27:
It looks to me that the issue is somewhere in graphics driver.

```
#0 0x00007ffff57e442f in () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/1 0x00007ffff5798a12 in raise () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/2 0x00007ffff578346d in abort () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/3 0x00007ffff57d8788 in () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/4 0x00007ffff5872a52 in __fortify_fail () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/5 0x00007ffff587296d in () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/6 0x00007ffff58728ce in __longjmp_chk () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/7 0x00007ffff707f745 in () at /usr/lib64/libcurl.so.4
https://github.com/prusa3d/PrusaSlicer/issues/8 0x00007ffff5798ab0 in () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/issues/9 0x00007ffff58cfef8 in () at /lib64/libc.so.6
https://github.com/prusa3d/PrusaSlicer/pull/10 0x00007fffbb5707be in () at /usr/lib64/dri/crocus_dri.so
https://github.com/prusa3d/PrusaSlicer/issues/11 0x00007fffbb081002 in () at /usr/lib64/dri/crocus_dri.so
https://github.com/prusa3d/PrusaSlicer/issues/12 0x00007fffbb08190e in () at /usr/lib64/dri/crocus_dri.so
https://github.com/prusa3d/PrusaSlicer/issues/13 0x00007fffbb084fd9 in () at /usr/lib64/dri/crocus_dri.so
https://github.com/prusa3d/PrusaSlicer/issues/14 0x00005555563681c6 in Slic3r::GUI::GLModel::send_to_gpu() (this=0x7fffffff8ac0)
```

The fishy call in this call stack is libcurl.so.4, it should not be called from the GPU driver. Likely the stack trace is not correct due to optimization.[

PyBonnetainNesterenko, 2022-06-28:
> It looks to me that the issue is somewhere in graphics driver.
> 
> ```
> The fishy call in this call stack is libcurl.so.4, it should not be called from the GPU driver. Likely the stack trace is not correct due to optimization.[

And 

> I think we might need debugging symbols (and ideally sources) for sys-libs/glibc, net-misc/curl, and media-libs/mesa too. Feels weird there's a _fortify_fail in there.

I'm far from the « proper » computer for some days. Will look into this (meaning trying to get some useful backtrace) next week.

PyBonnetainNesterenko, 2022-06-28:
> The fishy call in this call stack is libcurl.so.4, it should not be called from the GPU driver. Likely the stack trace is not correct due to optimization.[

Should I try to recompile stuff with -O0 ?

PyBonnetainNesterenko, 2022-07-03:
Hello,

Compile the latest (Gentoo) version of prusaslicer (2.6.0_pre20220620-r1). Got several SIGALRM, and the dreaded "longjump" error. Here is the backtrace.

[2022-07-03 20:56:45.852136] [0x00007ffff1c0b640] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Timeout was reached:
Connection timeout after 10000 ms
[Error 28]
[2022-07-03 20:56:45.855102] [0x00007fffa6a01640] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2`: HTTP 0, Timeout was reached:
Connection timeout after 10000 ms
[Error 28]
[2022-07-03 20:56:45.855138] [0x00007fffa6a01640] [error]   Downloading PrusaSlicer version file has failed:
Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2`: HTTP 0, Timeout was reached:
Connection timeout after 10000 ms
[Error 28]
[Thread 0x7fffa6a01640 (LWP 26509) exited]
*** longjmp causes uninitialized stack frame ***: terminated

Thread 1 "slic3r_main" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	      return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff57e448f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff5798a12 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff578346d in __GI_abort () at abort.c:79
#4  0x00007ffff57d8788 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff590a888 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:155
#5  0x00007ffff5872a52 in __GI___fortify_fail (msg=0x7ffff590a847 <longjmp_msg> "longjmp causes uninitialized stack frame") at fortify_fail.c:26
#6  0x00007ffff587296d in ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:105
#7  0x00007ffff58728ce in __longjmp_chk (env=env@entry=0x7ffff70e0a00 <curl_jmpenv>, val=val@entry=1) at ../setjmp/longjmp.c:39
#8  0x00007ffff707f745 in alarmfunc (sig=<optimized out>) at /usr/src/debug/net-misc/curl-7.83.1/curl-7.83.1/lib/hostip.c:784
#9  0x00007ffff5798ab0 in <signal handler called> () at /lib64/libc.so.6
#10 0x00007ffff585775f in __GI___poll (fds=0x55555981d400, nfds=4, timeout=14817) at ../sysdeps/unix/sysv/linux/poll.c:29
#11 0x00007ffff5a98f0e in  () at /usr/lib64/libglib-2.0.so.0
#12 0x00007ffff5a99263 in g_main_loop_run () at /usr/lib64/libglib-2.0.so.0
#13 0x00007ffff5e22bcd in gtk_main () at /usr/lib64/libgtk-3.so.0
#14 0x00007ffff69d8605 in wxGUIEventLoop::DoRun() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#15 0x00007ffff6e68ded in wxEventLoopBase::Run() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#16 0x00007ffff6e300d6 in wxAppConsoleBase::MainLoop() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#17 0x00007ffff6eba628 in wxEntry(int&, wchar_t**) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#18 0x0000555555f24476 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /usr/src/debug/media-gfx/prusaslicer-2.6.0_pre20220620-r1/src/slic3r/GUI/GUI_Init.cpp:56
#19 0x000055555580b5e4 in Slic3r::CLI::run(int, char**) (this=0x7fffffffd840, argc=1, argv=0x7fffffffda78) at /usr/src/debug/media-gfx/prusaslicer-2.6.0_pre20220620-r1/src/PrusaSlicer.cpp:616
#20 0x00005555557efe01 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/media-gfx/prusaslicer-2.6.0_pre20220620-r1/src/PrusaSlicer.cpp:875`

Does not look exactly as previously.

thesamesam, 2022-07-03:
So the jump is from a timeout in curl? Huh.

thesamesam, 2022-07-03:
1. Can folks hitting this show me `emerge -v1p net-misc/curl` (to see the flags)?
2. Yeah, I really think this is to do with the curl timeout being safe or not:
- https://bugzilla.redhat.com/show_bug.cgi?id=539809
- https://curl.se/mail/lib-2008-09/0197.html
- https://curl.se/mail/lib-2015-03/0136.html

MuninnAndHuginn, 2022-07-05:
Original USE flags:

`~ ❯❯❯ emerge -vlp net-misc/curl                                                                         ⏎

These are the packages that would be merged, in order:

Calculating dependencies... done!
[ebuild   R    ] net-misc/curl-7.83.1::gentoo  USE="ftp http2 imap ipv6 openssl pop3 progress-meter smtp ssl tftp -adns -alt-svc -brotli -gnutls -gopher -hsts -idn -kerberos -ldap -mbedtls (-nghttp3) -nss (-quiche) -rtmp -samba -ssh -sslv3 -static-libs -telnet -test -threads -verify-sig -zstd" ABI_X86="(64) -32 (-x32)" CURL_SSL="openssl -gnutls -mbedtls -nss" 0 KiB
`

Updated USE flags to add adns (based on curl/fedora reports above:
` env USE="adns" emerge -av net-misc/curl

These are the packages that would be merged, in order:

Calculating dependencies... done!
[ebuild   R    ] net-misc/curl-7.83.1::gentoo  USE="adns* ftp http2 imap ipv6 openssl pop3 progress-meter smtp ssl tftp -alt-svc -brotli -gnutls -gopher -hsts -idn -kerberos -ldap -mbedtls (-nghttp3) -nss (-quiche) -rtmp -samba -ssh -sslv3 -static-libs -telnet -test -threads -verify-sig -zstd" ABI_X86="(64) -32 (-x32)" CURL_SSL="openssl -gnutls -mbedtls -nss" 0 KiB
`

I am assuming simply recompiling curl with c-ares and trying to run prusa-slicer is enough to test this. It appears to work properly now.

bluecube, 2022-07-07:
This problem seems to be fixed downstream in `media-gfx/prusaslicer-2.6.0_pre20220601-r2`. It still feels like an error that prusaslicer compiles fine with wrongly configured curl and then crashes.

MuninnAndHuginn, 2022-07-07:
> 

The change was related to line 32. The addition of the adns requirement for curl.
net-misc/curl[adns]

This will require curl with async DNS support in order to compile/install prusaslicer. So I guess more of a dependency specification adjustment than anything.

lukasmatena, 2024-02-19:
Closing. If I understand it correctly, there was not a problem in PrusaSlicer.

closed 2024-02-19 by lukasmatena