prusa3d/PrusaSlicer #8397: App silently shuts down while trying to Export to just plugged SD card half times

Artyuha, 2022-06-11:
### Description of the bug

Sometimes (nearly every second time) app silently craches while trying to export to just plugged microSD card.
App should never crash silently. It should at least say if something happend before it die.
No matter if I press the Export to SD card button just right after it appears or if I wait for some time (even some minutes), app can crash anyway.
If I plug in the microSD card first and start the PrusaSlicer app after that then this issue never occures.

### Project file & How to reproduce

1. Star the PrusaSlicer app.
2. Open any stl file.
3. Generate GCode.
4. Plug in the microSD card via adapter into the USB port (directly or throught hub not matter).
5. Wait while the Export to the SD card button appear on the right of the Export GCode button.
6. Press the Export to the SD card button.
7. 50/50 app crashes without any message.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2 and some earlier stable releases

### Operating system

Win 10 Enterprise 21H2

### Printer model

Creality Ender 3

stepikovo, 2022-06-19:
We are not able to reproduce this issue. Have you tried a different SD card?

lukasmatena, 2024-02-19:
No response, closing.