prusa3d/PrusaSlicer #8399: Speed settings for Purge Tower

Bumblebiber, 2022-06-12:
Hi everyone, 

I'm a proud owner of a Palette 3 Pro since recently. Now that I've started to use Purge Towers, I've noticed there is no setting for speeds for those. They usually print super slow which results in prints taking forever.

I've worked with the substitution script to replace speeds with higher ones, but this can lead to other errors and you need to find out the print speed of the wipe tower before you can do that, which not everyone is capable of.

By doing so, I was able to reduce the print time of an only 2 colored print from 7 to 3 hours!

So that in my eyes is a very important setting that's missing (and probably not only for Palette users).

Cheers,
Benjamin

exec79, 2022-06-29:
im also a palette (2sp) user, use some non round numbers at speeds to avoid accidental matching, here is how mine looks. This works for me so far, but i would like to see speed option as well.

[image]
[image]

bombela, 2022-12-14:
Same issue here. Can you share how did you hack the speed/flowrate for the print tower?

lukasmatena, 2023-11-15:
Since [2.6.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc1), the wipe tower speed is determined from perimeter speed and infill speed, as mentioned in the change log. Closing.

closed 2023-11-15 by lukasmatena