prusa3d/PrusaSlicer #8400: PrusaSlicer not responding when launched on second screen

ElfyShort, 2022-06-13:
### Description of the bug

When PrusaSlicer starts on a second screen, it does not load all the way, and freezes in a "not respond" state.
I have been catching this issue since I started using the slicer (I believe it was version 2.3.1 or 2.3.2). 
The bug was preserved till current version 2.4.2.

The way I used to make it running again involved turning off the 2nd screen, and launching the Slicer, making it load normally on the main screen.

### Project file & How to reproduce

Launch slicer.
Drag on second screen.
Close slicer.
Launch slicer (to make sure it is starting on second screen).
Slicer does not load.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.3.2 - 2.4.2

### Operating system

win 10

### Printer model

Custom model

stepikovo, 2022-06-19:
Can you please check if this is the issue/solution? #5573

ElfyShort, 2022-06-19:
> Can you please check if this is the issue/solution? #5573

I have the same behavior on my home HP laptop and my office Fujitsu desktop. Both setups uses double monitors of the same resolution. Neither use Nahimic or any weird software. All the hardware (mouse/keyboard+drivers) are different. 
I have tried the 2.4.1 build suggested in the end of #5573, and that still shows the same behavior, when launched on second monitor:
[image]

lukasmatena, 2022-06-20:
Thanks for the report. Can you please try [the following build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/ys/PrusaSlicer-2.6.0-alpha0+147-ys-wx317-gtk3-debug-win64-g88bb6fc00-202206200748.zip), where we tried to fix the problem?

You may need to enter following credentials to download the build:
user: prusaslicer
password: slicer

ElfyShort, 2022-06-20:
> Thanks for the report. Can you please try [the following [...]

This build seam to be forcefully launching PS on main screen, no matter where it was closed.
It also always starts in a very small window and does not "remember" the window size I am setting.
Looks more like a work around than a fix, but if the window size will be "remembered", this can potentially work.
[image]

YuSanka, 2022-06-23:
Thanks, @ElfyShort. And can you please [one more build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/ys/PrusaSlicer-2.6.0-alpha0+156-ys-wx317-master-win64-g22aa2f9b7-202206221611.zip)?

As above, you may need to enter following credentials to download the build:
user: prusaslicer
password: slicer

ElfyShort, 2022-06-23:
> Thanks, @ElfyShort. And can you please try [one more build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/ys/PrusaSlicer-2.6.0-alpha0+156-ys-wx317-master-win64-g22aa2f9b7-202206221611.zip)?
> 
> As above, you may need to enter following credentials to download the build: user: prusaslicer password: slicer

I see no difference between this and the previous build I tested, so exactly the same feedback.

tschaerni, 2024-11-30:
2.5 Years later and this is still an issue.

closed 2026-09-29 by prusaslicer-bot[bot]