prusa3d/PrusaSlicer #8404: Export as STL off of the build plate results in a 0kb file

TKIY, 2022-06-14:
### Description of the bug

I was repairing STLs and had several to do.  Some were arranged off of the build plate.

When I export the repaired file, it comes out as 0/1KB and obviously corrupt.  The object is still fine in PrusaSlicer though.  If I move the object back on to the build plate and export again, the exported file is fine.

### Project file & How to reproduce

[BugReport.zip](https://github.com/prusa3d/PrusaSlicer/files/8902142/BugReport.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+Win64

### Operating system

Windows 10

### Printer model

Elegoo Saturn

foreachthing, 2022-06-15:
Well, not sure why, but when I hit "arrange" first, so your part is actually on the print plate, it exports fine (5.7 MB).
If I leave the part where it is (outside the print plate), it only exports 84 bytes... which is basically an empty file.

TKIY, 2022-06-15:
Yes the root issue is that exporting an object off the build plate creates a blank file without any warning.

I was working with about forty objects before I noticed this behaviour, and some were off the build plate.

If this is intended behaviour there should be some sort of dialog box to warn the user that the object won't actually be exported, or if it's unintended it's a bug.

TKIY, 2022-06-15:
No, I'm not using 'export plate'.  Right click on the object and choose 'export as STL'.

foreachthing, 2022-06-15:
OK! I missed that part!
Funny thing: Using your 3mf, export does not work 
Import your 3mf to my settings, move it outside the plate and export it - it works!

fake-name, 2022-10-11:
It also does this if your part is too big to fit on the plate. 

From what I can tell, if prusa slicer thinks a part isn't "printable", it also becomes un-exportable via any mean. 

This is super silly. I sometimes use prusaslicer to convert `.3mf` files to `.stl` (to feed into software that doesn't know how to consume `.3mf` files), and the fact that the estimated printability has *any* effect on whether the part gets exported is ridiculous.

closed 2026-09-29 by prusaslicer-bot[bot]