prusa3d/PrusaSlicer #8406: PrusaSlicer emits "G1 F0" which causes print to fail

parhelia, 2022-06-15:
### Description of the bug

PrusaSlicer emits "G1 F0"  which would cause the print to fail 100% of the time.

Workaround: manually remove all occurences of "G1 F0" in the GCCODE file.

Root cause: when trying to use advanced speed settings, users may enter a value of "0" for support, thinking that the slicer will algorthimically derive the maximum speed, even if supports are not used in the print. This will cause PrusaSlicer to generate "G1 F0", which will cause the print to fail.

### Project file & How to reproduce

[Tollerance%20Test%20v2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8911137/Tollerance.20Test.20v2.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+x64

### Operating system

macOS Montery 12.4 (21F79)

### Printer model

TriLAB DeltiQ 2, Controller Version: 1.1.4, Duet Version: 2.04trilab59

lukasmatena, 2022-06-16:
You are right, although it is quite a corner case. It only appears when you use autospeed here and define speeds for the rest of the features. In that case, the autospeed is not correctly resolved and the zero speed is used for skirt and brim (which are normally printed using support_material_speed). I just checked in four years old SlicePE 1.39 and the bug is already there, I assume it was there the whole time since autospeed was implemented.

dderbidge, 2023-08-12:
> You are right, although it is quite a corner case. It only appears when you use autospeed here and define speeds for the rest of the features. In that case, the autospeed is not correctly resolved and the zero speed is used for skirt and brim (which are normally printed using support_material_speed). I just checked in four years old SlicerPE 1.39 and the bug is already there, I assume it was there the whole time since autospeed was implemented.

This happened when I set "first layer speed" to 0. Maybe 0 isn't a valid setting for that field. If not, perhaps it could give a warning about that.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]