prusa3d/PrusaSlicer #8408: layer specific infill crashes

meteor42, 2022-06-16:
### Description of the bug

I want to change the infill in layers 11.0 - 35.0.
When i enter a first layer in dialog "Einstellungen für Höhenbereich" settings for layers,
and press <Enter> the slicer often crashes sometimes on the 2nd parameter (with <Enter>).
[project.zip](https://github.com/prusa3d/PrusaSlicer/files/8917755/project.zip)

### Project file & How to reproduce

open project spinningDiceBalanced
select object mySpinningDice8Alfa.stl (the cone without layer spec. settings)
select "Höhenbereich Modifizierer" layer region modify
Enter "13" in 1st Inputfield and press <Enter>
Crash - prusaslicer ends without message.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+linux-x64-GTK3-202204251120.AppImage

### Operating system

Linux Mint 20.2 Cinnamon

### Printer model

prusa mk3 s

lukasmatena, 2022-06-16:
Thanks. This is a Linux specific issue that we are already aware of. We will try to fix it.
@YuSanka FYI

lukasmatena, 2022-06-16:
Also, it seems to be already reported in #8055.

bubnikv, 2022-06-29:
Fixed in PrusaSlicer 2.5.0-alpha2.
Closing.

closed 2022-06-29 by bubnikv