prusa3d/PrusaSlicer #8409: Show totals on used filament in gcode previewer

jokohoko, 2022-06-16:
At the moment the gcode preview interface shows the time a print would take. And the material usage per feature type. But there is no total on the used filament.
[image]
I would like one extra line below with the total overview in meters and gram.

rtyr, 2022-06-17:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8208. Closing.

closed 2022-06-17 by rtyr

jokohoko, 2022-06-18:
Oh yeah, it's the same code behind both probably. I made a new request since it is missing completely in the gcode viewer. In the regular slicer the totals are visible in the bottom right. But adding it to both would be ideal indeed.

stevepowell2000, 2024-01-22:
There is movement on https://github.com/prusa3d/PrusaSlicer/issues/8208
Does the core team watch for PRs?