prusa3d/PrusaSlicer #8410: Bed Temperature other Layer not disabled

tome9111991, 2022-06-19:
### Description of the bug

When you set Bed temperature other layer to 0.
the setting will not disable you have a M140 S0 in Gcode

### Project file & How to reproduce

1. got to Filament Settings
2. Filament
3. Set Temperature Bed Other layers: 0
4. Slice 
5. Look Gcode you can find M140 S

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Win10

### Printer model

E3v2

tome9111991, 2022-06-20:
okay.
but if you set nozzle temp other layers to 0
you dont habe M104 S0

Kekht, 2022-10-31:
Confirming.
Behaviour for nozzle temperature:
Set "Other layers" = 0 will disable emitting M104 after first layer - as described in tooltip

Вehaviour for bed temperature:
Set "Other layers" = 0 will emit M140 S0 after first layer (tooltip describes that setting to 0 should disable g-code emitting after first layer).
But, if, for example, we set both "First layer" and "Other layers" to the same value, say 60, then:
If we have start g-code that contain "original value":
M109 S{first_layer_temperature[0]} -> M109 S60
Then after first layer, M140 will not be emitted (because command will be the same as for first layer)
But if we have calculated (for example conditionally)
M109 S{first_layer_temperature[0]+10} -> M109 S70
then M140 S60 will be emmited (with value from "Other layers" field) - because command differs from that at start

Kekht, 2022-10-31:
Tried to fix this by inserting g-code at "Before Layer Change" and "After Layer Change"field
`;LAYER_CHANGE
;Z:0.6
;HEIGHT:0.3
G1 Z.6 F9000
; == Try to temporarily fix a bug ==
	
M140 S70 ; set bed temp

M140 S60 ; set bed temperature
G10 ; retract`
It seems that buggy M140 emission occurs after that

scramsby, 2023-03-28:
Can confirm this bug is still present in 2.6.0-alpha5 and caused a few print failures for me until I figured out what was going on.

adolson, 2023-04-04:
Still an issue on 2.6.0-alpha6...

There is no reason this should be happening if the temperature is set to 0. Doubly so, if the printer setting "Emit temperature commands automatically" is unticked. Yet it does, in both cases.

> I think that is correct behavior if first layer temperature is set to non-zero value. But maybe we should improve the tooltip.

I disagree. "Set this to zero to disable bed temperature commands in the output" indicates it is not correct behaviour. And as someone else pointed out, it is not consistent with the nozzle temperature behaviour, so improving (read: completely rewriting) the tooltip doesn't make sense when the behaviour is problematic.

Kekht, 2024-01-14:
Still an issue on 2.7.1 :(
It still not working as description says
[image]

Kekht, 2024-09-26:
Still an issue on 2.8.1...

closed 2026-09-29 by prusaslicer-bot[bot]