prusa3d/PrusaSlicer #8411: Add command line arguement to change default

Norbs12, 2022-06-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

lukasmatena, 2022-06-20:
The `--datadir` parameter is available for years.

closed 2022-06-20 by lukasmatena

Norbs12, 2022-06-20:
> The `--datadir` parameter is available for years.

I swear I dug around for command line arguments in both slic3r and prusaslicer before posting this but looks like I overlooked it.

Thank you!