prusa3d/PrusaSlicer #8412: Brim Makes First Layer of Supports Disappear

djngnear, 2022-06-20:
### Description of the bug

Enabling an Outer Brim effects the first layer of support on the inside of an object. That first support layer is just gone and the support wants to keep building above it.

[image]

[Adapters and Caps.zip](https://github.com/prusa3d/PrusaSlicer/files/8936884/Adapters.and.Caps.zip)
.

### Project file & How to reproduce

So far this is only for an object with internal walls.
Enable supports and first layer expansion.
Enable outer only brim.
Slice and look for first layer of supports.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+win64

### Operating system

Windows 10

### Printer model

Prusa MK3s+

rtyr, 2022-06-20:
Reproduced, thanks.

puckpuck, 2022-12-16:
I just hit this same problem and it's causing me enough pain. I even tried adding a 0.2mm slab to my print and it just raised when the supports would start printing to 0.6mm causing the same problems.  

Need this fixed asap.

jamespearson04, 2023-05-07:
Encountering this problem on 2.6.0 too, makes brim and support mutually exclusive in a lot of cases.

Vermuden, 2026-02-22:
Is a fix planned for this? I encountered the same issue on version 2.9.4 (Flatpak) & Core One. It's very annoying as it makes anything with supports on PC/ASA impossible to print, as these require a brim to prevent warping.

closed 2026-09-29 by prusaslicer-bot[bot]