prusa3d/PrusaSlicer #8413: Adding custom G-Code from UI sets the Color Print view

dst74, 2022-06-20:
### Description of the bug

Adding custom G-code from the UI slider sets the view in Color Print
[image]
.

### Project file & How to reproduce

Click on + on the cursor
Add, for instance, a print pause

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+linux-x64-GTK3-202204251120

### Operating system

Ubuntu 22.04 LTS

### Printer model

MK3S+

rtyr, 2022-06-20:
It is intended behavior. It is not meant only for color print, you can also visualy see the pause and see the estimate time in the legend.

lukasmatena, 2023-02-01:
The behavior was changed after all in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena

julian-weinert, 2023-09-26:
Good it got changed. Because it doesn't make any sense whatsoever.
Is this behaviour changed when adding a pause as well?