prusa3d/PrusaSlicer #8415: Allow print multiple sequential layers between tool changes

ardenpm, 2022-06-20:
**Is your feature request related to a problem? Please describe.**
When doing multiple colour/multiple material printing, regardless of the mechanism used (tool changer, multiple-in-one-out, IDEX etc), the number of tool changes and total time spent changing tools can dominate the printing time, so anything that can reduce the number of tool changes can significantly improve print times.

**Describe the solution you'd like**
The ability to allow sequential printing of multiple layers (of a configurable number) of objects assigned to a given tools before performing a tool change operation.

So, for example this would allow me to print say 4 layers using the current tool, then change to the next tool and print the same four layers of the objects assigned to that next tool.

The exact implementation would be a little difference since of course PrusaSlicer always prints the next layer based on the current tool it already has selected, so this staircasing approach could still remain, it would just be doing more layers in between.

Obviously like regular sequential printing this would introduce clearance issues and as with that feature some warning should possibly be given. It would be quite model dependent, for example if the number were too high and there was a steep overhang it would obviously create issues, however for a large number of multiple colour/material models it could be effective.

The nozzle would need to be raised during travel to ensure the layers from the previous tool are cleared. When going this manually in Gcode I have used the Z-hop functionality to achieve that but ideally it should be automatic based on the chosen number of layers.

I don't believe this issue should prevent the addition of this feature since PrusaSlicer already contains other functionality which is quite model dependent and requires care on the part of the user to ensure it is appropriate to their specific model.

If a prime tower is used then it would need to be kept in sync, the last tool for the sequence could add the needed number of layers to the prime tower. This should make the approach compatible with both tool changer and MMU approaches.

**Describe how it would work**
Ideally a new (likely advanced) option would be added to the multiple extruders print settings which allows the user to specify a number of layers to combine tool changes for.

Path planning would then be changed to work similar to sequential printing (but obviously not with XY clearances) but within the number of layers specified in a repeating cycle. The Z height to set during travel needed to clear those layers should be automatically computed.

**Describe alternatives you've considered**
G-code post processing, however this is extremely complex given the layers would need to be interleaved at the same height and travel moves would already have been baked in.

**Additional context**
This concept was mentioned by Andy Cohen on the [3D Printing Today](https://podcasts.apple.com/us/podcast/3d-printing-today/id821012321) podcast as being used in another slicer to reduce the total number of tool changes quite dramatically within a print.

Potentially depending on how many layers you can get away with tool changes could be reduced by factors of 2 - 4 pretty safely for many models and print times quite dramatically along with it.

Being able to vary this setting by layer height with a layer height modifier would also be valuable so that if a model contained a small section which would pose a problem for this method it could be handled.

I searched for other similar issues and feature requests but was not able to locate one, possible because terminology used in this area can vary quite a lot.

JohnOCFII, 2022-06-21:
It is a fascinating idea.  I too heard about it on that podcast.

nomike, 2024-11-18:
I just received my MMU3 and I'm doing my third print on it and just had the same idea.

To make it a bit simpler and add to it:

During regular single-tool, single-material printing, the printer is printing one layer after the other. But there might already be obstructions.

Let's say it wants to print a hollow cylinder. So first, the perimeters are extruded. Next to be extruded will be the infill. On any non-trivial geometry, it might happen, that during printing, the print head has to travel a few times to different spots of the model. In order to not crash into already extruded perimeters, infill. etc. the head needs to be slightly lifted up on t z-axis, it travels to the destination, if lowered on the z-axis again, extrusion can continue.

So this part is already built into the slicer. I assume, that there is  fixed factor (e.g. 2 x layerheight or something) for these z-axis travels.

To implement this feature. the following would need to be changed:

1. A new parameter `combine_toolchange_every_n_layers` needs to be added, which has a default value of 1 and could be set to 2, 3, 4 (I guess, there will be a limit of where it's no longer feasible. Like at 4 it might no longer only be the nozzle-tip which needs clearance but other parts from the extruder as well (heat-block, fan duct, etc.).

2. The internal calculation of the z-axis travel needs to be `z_axis_travel_height = z_axis_travel_height + (combine_toolchange_every_n_layers * layer_height)` (probably slightly more complex to accommodate for variable layer height, but you get the idea).

3. Follow this algorithm:

```python
for tool in list(set([i.tool for  i in current_layer.all_extrusions])):
    for i in range(0, combine_toolchange_every_n_layers):
        set_z_axis(layers[current_layer].z_start + i * layer_height)
        slice_path_for_tool(layers[current_layers], tool)
```

And for all travel moves inside `slice_path_for_tool` the z-axis travel needs to be calculated using the formula I mentioned above (` z_axis_travel_height = z_axis_travel_height + (combine_toolchange_every_n_layers * layer_height)`.

I don't have the energy and time-resources to get familiar with the source code of the prusa slicer, but it doesn't seem all too complicated, and the benefit could easily be a reduction of print times for MMU prints of hours or by a factor of 2-5 depending on the model.

And less tool changes also means less wasted purge material.

And now that I think of it, the `No sparse layers` feature is not that vastly different from this idea.

nomike, 2024-11-18:
Oh, and the "combine infill" feature should already have everything implemented which is needed for this.

5jvm0u4, 2025-02-21:
Now with the new sequential algorism from 2.9.1 alpha this should become more viable to implement (?

closed 2026-09-29 by prusaslicer-bot[bot]