prusa3d/PrusaSlicer #8418: Allow changing of default Preview tab 'Show' Options

piebaker, 2022-06-21:
**Is your feature request related to a problem? Please describe.**
It would be really helpful to change the default view and show options in the preview tab. For example, to always have the 'Seams' option on, rather than having to re-enable it every instance. 

**Describe the solution you'd like**
Allow for an ini or GUI method to change the view and show defaults in the preview tab

**Describe how it would work**
Either the user could change ini value's to correspond with the new default, or in the preferences menu under GUI there could be options for view/show defaults

**Describe alternatives you've considered**
See above, ini or GUI.

**Additional context**
N/A

dartrax, 2022-07-09:
This is somehow the generalized Version of FR #7434.

fortinmike, 2022-07-14:
This would be really useful to help remember to tweak some settings such as seam positioning, avoid crossing perimeters (travel, for less PETG stringing), etc. But for me it's the seams option that I'd love to keep ON by default the most, as it has the most impact on part finish. I came to write an issue and found this one.

Combinacijus, 2022-11-04:
Did some digging in the code but I don't really understand it because the code base is very big and complex for me. So if someone who's familiar with code can give some guidance/examples on how one would approach saving.loading settings `std::string options_items` would be appreciated

Currently those settings seems hardcoded <https://github.com/prusa3d/PrusaSlicer/blob/4dee789e9e8aa55e3d9e9c03619767b500b74d18/src/slic3r/GUI/GUI_Preview.cpp#L264-L277>

To save settings **maybe** `wxGetApp().app_config->set(); wxGetApp().app_config->save();` are used
<https://github.com/prusa3d/PrusaSlicer/blob/27a18065b3c7af5d78fad8890d591f83d1a22194/src/slic3r/GUI/MainFrame.cpp#L1186>
<https://github.com/prusa3d/PrusaSlicer/blob/d81009c6ce1c823570d511905455e5a0e6416638/src/slic3r/GUI/Preferences.cpp#L842-L843>

piebaker, 2026-09-02:
Discussion opened ref [15655](https://github.com/prusa3d/PrusaSlicer/discussions/15655) due to automated cleanup as requested

closed 2026-09-02 by piebaker