prusa3d/PrusaSlicer #8419: Ugly printing on top surface

laptop-laptop, 2022-06-22:
**Is your feature request related to a problem? Please describe.**

Hi, in my CAD model, the surface looks smooth. However, after importing the STL file to PrusaSlicer and sliced, 
some yellow scaled-like things (highlighted in blue) shows up on the surface in PS. After printing, the hightlighted area looks very ugly.
Could you please tell me what is wrong and how to fix it?

[img]https://i.imgur.com/EqL0gOy.jpg[/img]
[img]https://i.imgur.com/K49Jvtj.jpg[/img]

Sorry model file cannot be posted.

rtyr, 2022-06-22:
Anwered. Closing.

closed 2022-06-22 by rtyr

laptop-laptop, 2022-06-22:
Thanks @foreachthing 

By layer height, do you mean reducing the default layer height of 0.15mm when printing that region? The thickness of the surface is 1.5mm.

JohnOCFII, 2022-06-22:
> Why I need to do this now? Under what situation do I have to do it?

It has to do with that part being more horizontal than vertical.  The layer lines get spread out in that case.  The only remedy is to increase the number of layers in those areas.  This would also be a good opportunity for you to try variable layer height.  https://help.prusa3d.com/article/variable-layer-height-function_1750

laptop-laptop, 2022-06-22:
Understood. Thanks. Increasing the print time due to smaller layer height and nozzle diameter is a problem for large prints though. I guess using Adaptive layering might reduce the time only slightly.