prusa3d/PrusaSlicer #8420: Scale object between two points defined by the user

mtb2430, 2022-06-23:
I'd like to suggest a feature relating to scaling/resizing of models.
The current user interface appears to only scale the models size according to its overall dimensions, or as a percentage of its current size. My proposal is to be able to resize a model between 2 points on the model defined by the user.

I'll use my scenario as an example of where this would be a valuable feature, and I'm sure this would also be beneficial to many other users with their own scenarios. I wish to scale a model(s) of a car body to suit an existing R/C car(s). The most accurate way to do this would be if the user were able to define 2 reference points on the model (i.e. the centre line of the front wheel opening, and centre line of the rear wheel opening) and input the distance they require the model to be sized between these 2 points in order to suit an existing R/C car wheelbase.

lukasmatena, 2022-06-23:
This is a duplicate of #8318, which itself is a duplicate of #3651. I suggest we close this a discuss in #3651, if there is something left to discuss.

lukasmatena, 2022-11-16:
Closing, duplicate of #3651.

closed 2022-11-16 by lukasmatena