prusa3d/PrusaSlicer #8421: SL1S time estimate incorrect for slower tilting

murk-sy, 2022-06-23:
### Description of the bug

When slicing a file in a custom material, I noticed that the estimate was 1h15m in the slicer, however on the printer it immediately showed an estimate of 2 hours.

Further investigation revealed that selecting slow tilt does not affect time estimate.

### Project file & How to reproduce

1. Select 0.05 Normal, Prusament Resin Tough Prusa Orange, and Original Prusa SL1S SPEED settings
2. Create a 10 mm box and disable supports and pad (this is for time reproduction)
3. Slice
4. Time estimate is 18m 1s
5. Under Material Settings > Material printing profile > Print speed, instead of Fast, select Slow
6. Time estimate is incorrect and you can't manually reslice.
7. Change support settings back and forth so you can reslice
8. Time estimate is 18m 1s, even though it should be 500 seconds longer (2.5s fast-slow difference * 200 layers), so 26m 21s

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10 x64

### Printer model

SL1S

stepikovo, 2022-06-27:
This should be fixed in 2.5.0

bubnikv, 2022-06-29:
Implemented in PrusaSlicer 2.5.0-alpha2. Closing.

closed 2022-06-29 by bubnikv