prusa3d/PrusaSlicer #8423: Make first layer of top solid infill anchor to the nearest internal infill line beneath it.

boksbox, 2022-06-26:
**Is your feature request related to a problem? Please describe.**

If the model has a single top surface, e.g. flat disk or cube, the first layer of the top solid infill anchors to the perimeter of the model.  That first layer of the top solid infill also slows down and works like a bridge.
However, when there are more than one top surface, e.g. a logo,  the first layer of whatever lower top surface will begin to be extrude on to the internal infill beneath it based ot the shape of the design it is printing, e.g. a letter or design of a logo or generally any model that has several top surfaces.
 
Now since the slicer blindily follows the slightly enlarged image of the design above it, the start (or finish) of the layer lines that "float" on mid-air (portions of the internal infill that has empty space) has nothing to anchor to  thereby causing fraying and snags that mess up the print.

In my screenshot attached, the edges of the circle does not anchor to anything, causing fraying and generally bad surface finish.
Monotonic infill does not work here.

**Describe the solution you'd like**

Since the first layer of the top solid infill should be a bridge, it should not merely be an enlarged image of the design where start and end of the bridge have nothing to anchor to.  What I propose is, that the image should be "pixelated" or its edges should be right on top of the (internal) infill pattern beneath it so the extruded lines have some edges to anchor to.  As much as posible, the lines of this first top solid infill should be not be parallel to the infill lines beneath it.

On the image below, the edges of the circle should be rough "pixelated" squarish edges are so they are right on top of the infill pattern beneath it.  That should take care of printing the begining of the bridge line having something to anchor to.

**Describe how it would work**
no programming skills.  Hoping someone to here would pickup and implement this based ot the solution above.

**Describe alternatives you've considered**
-none-

**Additional context**
Add any other context or paint drawing about the feature request here.
[image]
[image]
[image]

fredconex, 2022-07-23:
I vote for this feature, a perimeter around bridge infill would help on this kind of anchoring, bridge infill is being printed mid air, having a perimeter around it would fix it, Cura actually have this perimeter and I have no problems with top surface quality but with SuperSlicer/Prusa there's failures on top layer caused by this lack of anchoring.
[image]
[image]

stepikovo, 2022-07-26:
I'm closing this as a duplicate of #569

closed 2022-07-26 by stepikovo