prusa3d/PrusaSlicer #8424: Unwanted support

tonycstech, 2022-06-26:
### Description of the bug

Support is generated outside of the model.
Enforcing support only inside the model does not help.
Blocking support outside the model does not help.
This has been an issue for years.
It only goes away if i make support spacing smaller, but i do not want to do that. I want to make less support.

Also, support is generated on the side of the model not supporting anything.
Also support is generated above the model, beyond built plate, above already finalized support.

[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[Plug.zip](https://github.com/prusa3d/PrusaSlicer/files/8985970/Plug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

win10x64

### Printer model

na

jmd73, 2022-06-27:
Try changing support style from Grid to Snug.

stepikovo, 2022-06-27:
Unfortunatley your 3MF does not include supports, however, as jmd73 mentioned, you can try Snug supports.

tonycstech, 2022-06-27:
I need grid. This is a very anoting bug and I am surprised it existed for
so long.

stepikovo, 2022-07-02:
@tonycstech can you please provide us with the 3MF file?

tonycstech, 2022-07-03:
[model.zip](https://github.com/prusa3d/PrusaSlicer/files/9034122/model.zip)

stepikovo, 2022-07-07:
It's really hard to help when you don't provide a file from the screenshots. When I tried to block supports on your file, it worked just fine. The easiest way how to get rid of the "spilled" support structures is to use Snug, that's why it was created

tonycstech, 2022-07-07:
I dont have that model anymore, it was a temporary design.
Once printed, no longer needed.
I have the issue of support outside the model boundaries allot.
Not every model, but some that are more or less complex.

rtyr, 2022-07-07:
> I have the issue of support outside the model boundaries allot.

Please provide some examples.

tonycstech, 2022-07-07:
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/9061994/test.zip)
there you have it
10mm spacing for support will make it stick outside of the model creating unwanted patterns.
This is not the worst case, ive see even more weird stuff like that. Ive seen support generated above support, even when it said platform only.

rtyr, 2022-07-07:
You are using grid supports, therefore support columns are projected into a regular grid (it doesn't follow exactly the model shape). It is more stable but it also has some downsides. It can be possibly improved, but you should use snug support style in this case wich such huge spacing.

> No other slicer has this issue.
Any particular slicer you want to mention?

tonycstech, 2022-07-07:
cura
slic3r
craftware
repetier
pretty much all of them as far as i can tell.
none of them were generating supports in on the side of the model
[image]
Looking at the image, i cant wrap my head around what does it support, air ?
I understand platform for support to be slightly larger then the support its self, but larger by how much, by support spacing ?
Weird.
I can see an option to change that would be helpful. that value should be 0 by default so it does not happen.

Sometimes i get chucks in mid air outside of the model just hanging and supporting absolutely nothing.
Ill see if i can get any examples.

rtyr, 2022-07-07:
The behavior on the screenshot (with huge 10mm spacing) is certainly not ideal and should be improved, but you can easily solve this now by switching to snug supports.

> I can see an option to change that would be helpful. that value should be 0 by default so it does not happen.

Switch to snug supports. Grid supports are projected to a grid.

> cura
> slic3r
> craftware
> repetier
> pretty much all of them as far as i can tell.

As for slicers you mentioned. Cura uses snug supports only - you can use them too in PS to prevent issues related to grid support type. Not sure about Craftware support generator, others are not so relevant to be honest (and Repetier is not a slicer, it uses cura/slicer slicing engine). I tried your object in Simplify3D and it also had some bleeding issues with large support pillar resolution value (low density grid), Same in Ideamaker.

[image]
[image]

> Sometimes i get chucks in mid air outside of the model just hanging and supporting absolutely nothing.
> Ill see if i can get any examples.

Please search for similar issues first and if you can't find anything similar, open new issue. It is not good to mix different problems in one issue.

closed 2026-09-29 by prusaslicer-bot[bot]