prusa3d/PrusaSlicer #8426: Need a placeholder {total_weight_Including_spool}

EnoSolution, 2022-06-26:
Hello,

To know if there is enough filament left on the spool for the next print, I think the easiest way would be to have a {total_weight_including_spool} parameter. 

By including this parameter in the gcode file name, it would be enough to weigh the spool before printing and check if its weight is greater than the {total_weight_including_spool} of the object.

I know there is a QRCode on the prusament spool to do this check, but this placeholder would be a solution for all sppols from all manufacturers.

Best regards,

Enos

foreachthing, 2022-06-26:
There are already a bunch of placeholders available (https://help.prusa3d.com/article/list-of-placeholders_205643):
* total_weight
* used_filament
* extruded_volume

I guess you could weigh the spool when it's full and take note of that.

foreachthing, 2022-06-27:
> from an almost empty spool

I just count the revolutions of filament and measure (roughly) the inner diameter of the spool. That has been accurate enough - so far.

EnoSolution, 2022-06-27:
Okay foreachthing, I only just realized that you were kiding me :-)

Thanks stepikovo!

bubnikv, 2022-06-29:
You may enter the following formula

```
{total_weight+filament_spool_weight[0]}
```

into your output file name template to generate weight including the spool. Just save it into your profile and you don't need to repeat it again.

EnoSolution, 2022-06-29:
Thank you, it works very well. 

Now, if I have any doubt, I can check if there is enough filament left for printing by weighing the spool before printing (which can be done without unloading the filament) and making sure that the weight indicated is greater than the weight mentioned in the name of the object to be printed. No calculation required. Very useful.

Don't forget to update the list https://help.prusa3d.com/article/list-of-placeholders_205643

closed 2022-06-29 by EnoSolution

lukasmatena, 2022-06-29:
@EnoSolution 
>Don't forget to update the list https://help.prusa3d.com/article/list-of-placeholders_205643

There should be no need. The `total_weight` is mentioned and `filament_spool_weight` is a configuration option, so it is covered at the very top where it says `Each configuration option can be used in all of your custom G-Codes and output filename template.`

Is there something else missing?

EnoSolution, 2022-06-29:
Sorry, I didn't understand that all parameters were usable and that placing the mouse cursor over the input field displayed the name of the parameter to use.

Congratulations to the team!

foreachthing, 2022-06-30:
> There should be no need. The `total_weight` is mentioned and `filament_spool_weight` is a configuration option, so it is covered at the very top where it says `Each configuration option can be used in all of your custom G-Codes and output filename template.`

What am I missing?

```
!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 3: Not a variable name
{total_weight+filament_spool_weight[0]}
 ^
!!!!! End of an error report for the custom G-code template start_gcode
```

and this

```
Failed processing of the output_filename_format template.
Parsing error at line 1. Expecting a legacy variable expansion format
[input_filename_base]_[{total_weight+filament_spool_weight[0]}].gcode
                       ^
```

EnoSolution, 2022-06-30:
@foreachthing 

Try this syntax:

{input_filename_base}_{total_weight+filament_spool_weight[0]}.gcode

If you prefer, you can hide unusefull decimal digits in the weight with this syntax:

{input_filename_base}_{int(total_weight+filament_spool_weight[0])+1}g.gcode

bubnikv, 2022-10-11:
There is a placeholder `filament_spool_weight` however please note that it
is a per extruder vector variable.