prusa3d/PrusaSlicer #8429: Wall bulge when wall is 3 perimeters by design

rinkek, 2022-06-28:
### Description of the bug

When printing walls which are 3 perimeters wide by design, the wall starts to bulge and is not dimensionally accurate.
This works fine when the wall is 2 perimeters wide.
Also the seam becomes very ugly.
This all works perfectly fine when using Cura.
[PrusaSlicer Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/8999729/PrusaSlicer.Bug.zip)

I uploaded a ZIP with examples.

### Project file & How to reproduce

Just print the HollowRoundBox_3_wall.stl file, also added as .3mf or gcode. And you will see it's not nice.
When printing the Cura gcode, you'll see it's perfect.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa Mini+

rinkek, 2022-06-28:
Ow, and i turned off "Fill gaps", didn't help.

NGC869, 2022-06-29:
Hi,
this is a long know problem and has been previously pointed out here: https://github.com/prusa3d/PrusaSlicer/issues/7973

rinkek, 2022-06-29:
> Hi, this is a long know problem and has been previously pointed out here: #7973

Good to know, but also sad it's not solved already. There are quite a few people with ugly prints out there.
Of of the many topics: [https://forum.prusa3d.com/forum/prusaslicer/mitigate-wall-bulge/](url)

closed 2022-07-02 by stepikovo