prusa3d/PrusaSlicer #8431: wtf

Nachowarrior, 2022-06-29:
### Description of the bug

i have to sort through 30 menu's to change my nozzle diameter when it used to be easily accessible. now i have to create 90 different profiles for every fucking detail when i just want to throw off a one off print. disgusting. want to change your nozzle diameter? sure, just set up an entirely new printer.   fuck off. 

### Project file & How to reproduce

what the literal fuck. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

the shitty version.

### Operating system

i operate on  your mothers compliments

### Printer model

literally doesn't fucking matter

Nachowarrior, 2022-06-29:
same fucking printer, same everything. but now i have to type in the ip address every god damn time rather than setting it ONCE like it used to be.  why the fuck do i need to manually enter the ip address for every fucking profile when i used to be able just to set the mother fucker once?

Nachowarrior, 2022-06-29:
thank you for destroying my workflow on all of my previous profiles.

bubnikv, 2022-06-29:
> i have to sort through 30 menu's to change my nozzle diameter when it used to be easily accessible. 

What versions of what software are you comparing? Frankly I am not sure what software change you are referring to. Changing nozzle diameter always was being done in printer profile in Slic3r / Slic3r PE / PrusaSlicer.

> same fucking printer, same everything. but now i have to type in the ip address every god damn time rather than setting it ONCE like it used to be. why the fuck do i need to manually enter the ip address for every fucking profile when i used to be able just to set the mother fucker once?

The IP address used to be part of the Printer profile, however since PrusaSlicer 2.3.0 released on Jan 11, 2021, the IP address of a printer is set in a physical printer configuration. A single physical printer configuration may be linked with multiple printer profiles, you don't need to create a new physical printer for each printer profile.

May I please ask you to stop using profane language and act respectfully?

rtyr, 2022-06-29:
OP deleted his issue. Closing.

closed 2022-06-29 by rtyr