prusa3d/PrusaSlicer #8433: Slicer should generate a warning when it can see that printing is too slow (mostly during ironing) and clog is very likely to happen.

TomaszSzt, 2022-06-29:
## Is your feature request related to a problem? Please describe.

Whenever I enable ironing I have to always scratch my head: will it clog or will it not? I would be glad to have some support from a slicer to let me decide if to enable ironing or not. It would be also great for it to do something about it, but a warning will be sufficient.

## Describe the solution you'd like

Slicer should warn when it detects that there is a high probability that print will stop due to clogging created by "heat creep".

## Describe how it would work

### Mechanism of clogging due to "heat creep"

Since there are many contradictory information on the net what is the mechanism behind the "heat creep" I will try to explain how I do understand it and what experiments can be used to confirm that understanding.

The issue with clogging or "heat creep" is, in my opinion, specific to "minimum continuous flow rate" which is a characteristic of a printer, material and temperature. It is most visible during ironing but can be replicated if You will **slow down** the print too much. The effect is usually repetitive with a great accuracy, but very printer specific.

The hot end is basically a small vat of molten plastic. To push it out in a predictable way one must create some pressure. To do it the printer plunges in the non-melted filament which behaves like a piston in a cylinder.  Usually when person thinks about a piston and cylinder a person thinks about something like this:

[image]

The filament in a printer is however acting rather like "plunger actuator":

[image]

When a plunger submerges into the liquid it occupies some space and the liquid must be removed from the cylinder.

A printer filament extrusion is very like the plunger type pump, but with some peculiar and in fact ingenious changes. The tricky part is, that there is no gasket around this "filament plunger". There is however small, 0.1 or alike gap where air can escape, what is an absolute must, but this is also a space for a liquid plastic to climb up.

This climb of liquid plastic continues until the friction created by this self made "liquid gasket" will allow for enough pressure to build up in a melt chamber to actually push the molten plastic out of the nozzle.

This mechanism has two border line conditions:
 1. When extrusion speed is too high the level of plastic drops so much that non-melted filament starts directly touching the bottom of the nozzle. It can't build any hydraulic pressure in this case and must be physically pushed out through the hole in a nozzle.
 2. When extrusion speed is zero the piece of filament which was pushed into a vat of molten plastic melts. Due to this melting there is no longer a liquid gasket around the filament and there is no solid "plunger" below the surface of a melt. When filament starts to move again it has to submerge a bit to create this "liquid gasket" again. Until this gasket is created there is not enough pressure in melt chamber to extrude anything. Of course to make this gasket some liquid must *climb up* along the filament, like on below drawing:

[image]

The second borderline condition is responsible for:
  1.The inability of a printer to pause for a longer time and restart immediately right at the spot. There is always a small length which is not extruded correctly. One must use some "Extra length on restart" to compensate for it.
  2. The "heat creep" or "clogging" which happens:
   - during ironing of large are with low temperature plastic;
   - during very slow printing.

This happens, in my understanding, when filament is melting faster than it is being pushed out. In effect the "liquid gasket" keeps rising and rising up to the moment when it enters a cold zone (5). Depending on a printer, friction in cold zone, the actual size of a gap (depends on piece of nozzle and filament) and etc. it may be possible to push it down for a while or it will just jam and block the filament movement.

### Algorithm to detect it

I suppose we can assume a certain model of "melting rate" and we have known "feed rate" of filament. This may be used to create mechanism do detect the possibility of clog, but I suppose it will contain to many variables which can't be measured and will be hard to figure out for a person which is creating a print profile. 

I think it can be done with much simpler approach: 
- add Ve="minimum volumetric speed to warrant no clogging" option in filament profile, 
- add Vo="maximum extrusion volume which can be extruded below Ve to not clog" option in filament profile.
- add Vr="minimum extrusion volume above Ve required to prevent clogging" option in filament profile.

_it may be easier to understand and compute it if w will use "filament extrusion length" in GUI instead of "volumetric speed"_

Let Vacc be a variable to accumulate extruded problematic volumes along segments of generated extrusion paths.
Let dV(i) will be the volume to be extruded in current segment.
Let V(i) will be the volumetric extrusion speed of current segment.

```
If V(i)<Ve
  we have a segment which may contribute to clogging
if V(i)>Ve 
  we have a segment which contribute to un-clogging and a safe extrusion.

if V(i)<Ve  
{
   if   Vacc >= Vo         Warnings should be displayed to user that clogging is very likely to happen.
     else
    Vacc+=dV(i); 
}
if V(i)>Ve 
{
       Vacc-=dV(i);
       if (Vacc < Vr) Vacc = 0 // there is enough high speed extrusion to remove possibility of clogging.
}
```

The algorithm is obviously not fool proof and may have false positives, plus parameters will depend on a piece of a printer and will require fine tuning by user, but it is better to have some warning than none.

I think this algorithm should allow me to save on trial-and-error approach and modify design in such a way, that there is always something what is printed without ironing to clear the possibility of clogging.

rtyr, 2022-07-07:
Heat creep issues are affected by many factors, like hotend type, hotend cooling, filament type (and brand), ambient temperature, print temperature, ...

For example PTFE lined hotends doesn't suffer from such issues (see for example E3D lite6 hotend). It also happen mostly with PLA filament. Many users are enabling ironing by default in their profiles (mostly for printers with PTFElined hotends, like Creality printers).

Such warning would have to be configurable per printer/filament. Maybe we could just show some generic notification on the plater area whenever ironing is enabled, but I can imagine many users would find it annoying (especially when ironing is enabled by default).

starthal, 2024-03-26:
Assume you have a good heuristic for the maximum amount of area it's safe to iron with a given filament at a given flow rate, feed rate, and temperature.
Is it feasible to split a surface into patches and print them piecemeal so that you never exceed this time?
E.g. if you know you can iron 100mm^2 without risk, then:
- Print 100mm^2 of the top infill (stop at the nearest perimeter)
- Iron this patch
- Print the next 100mm^2
- Iron
- ... (Repeat until the surface is complete)
- Finally perform one ironing pass on each of the seams between patches.

I suspect this could be prototyped by reordering gcode in a script.

closed 2026-09-29 by prusaslicer-bot[bot]