prusa3d/PrusaSlicer #8435: Divide by Zero error upon slicing

guaddy, 2022-06-29:
### Description of the bug

When attempting to slicer any object with lightning infill set to any infill percentage, an "Divide by Zero" error shows up, stating to restart PrusaSlicer. After restarting PrusaSlicer, the error remains.

### Project file & How to reproduce

[test_item.zip](https://github.com/prusa3d/PrusaSlicer/files/9012122/test_item.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2

### Operating system

Windows 10 Enterprise

### Printer model

Prusa i3 MK3S+

lukasmatena, 2022-06-29:
@guaddy Thanks, reproduced. You are using automatic extrusion width (set by entering zero in `Print Settings->Advanced->Extrusion width->Infill`). Apparently this is not handled correctly. We will fix it.

guaddy, 2022-06-29:
@lukasmatena I can confirm that this is the issue. Changing the extrusion widths to defined values makes the lightning infill work properly.

closed 2022-06-29 by hejllukas