prusa3d/PrusaSlicer #8439: Inches units and scaling not working.  It changes values if clicked into boxes multiple times!

kevinmichaeldudley, 2022-06-29:
### Description of the bug

If you set the OBJECT manipulation to INCHES and enter a scaling of say 100.25 the inch values change.
Note: Only displaying 2 decimals.
But THEN if you pick into the inches box the SCALING changes on it's own to an arbitrary value.
If you continue JUST to click between boxes the values  CHANGE!

This does NOT happen if set to NOT INCHES (Metric/mm).

The displayed INCH values are rounded to 2 decimals.  That seems to be the problem.  It is loosing precision and rounding causes CONTINUALLY CHANGING values, at least when set to IN  (inches).

### Project file & How to reproduce

Just open any project and set the OBJECT manipulation to IN.
Change the scale to 100.25  (uniformly)
Click into the IN value.
Scale changes.
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 11

### Printer model

Mk3s and Createbot

kevinmichaeldudley, 2022-09-04:
FYI: Not fixed in 2.5.0 Beta RC1

kevinmichaeldudley, 2022-09-04:
Retest method:
Open project.
Set to IN (inches)
Click into any INCH value.
Now click into any scale of 100% and it CHANGES to a NEW value.

YuSanka, 2022-09-05:
Yes, this fix will be a part of next(2.6.0) release.

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena