prusa3d/PrusaSlicer #8441: Make "Avoid crossing perimeters" more intelligent and quality focused!

NGC869, 2022-06-30:
Hi,
PrusaSlicer still adopt some print strategies that ruin some kind of prints due to **unwanted nozzle jumps**.

In the image you can see a 0,42 mm thick ribbon printable with a single nozzle pass (with 0,4 mm nozzle). While the new Arachne perimeter generator facilitates this job, it still impossible to avoid the non-print movements that lead to bad stringing, especially when using rubber filaments.

This behavior is probably implemented to let the layer cool down, but truly often this feature is not needed and in fact introduce **bad stringing** in scenarios like the depicted one!

[image]

Following you can see the same print while **setting to zero the "Max detour length**" parameter: the long jump is avoided but PE introduce an equally useless non-print moves along the entire perimeter.

[image]

Following you can see the clean result obtained by **Simplify3D**:

[image]

Here attached the test STL:

[wave_thick042.zip](https://github.com/prusa3d/PrusaSlicer/files/9018277/wave_thick042.zip)

andy5macht, 2022-07-21:
I think what Simplify3D does is changing printing direction for each layer. So no travel is needed here.

NGC869, 2022-07-25:
> I think what Simplify3D does is changing printing direction for each layer. So no travel is needed here.

Hi Andy,
I think you are right, but there are also other dynamics involved.
Look at the attached example, a lattice composed of a double walls hexagons units: at the end of the external wall deposition the slicer move the nozze at the opposite area of the prints, and in my opinion this is a truly unintelligent choiche.

In this example that long "jump" has an hexagon shape due to the "avoid crossing perimeters" option setted to "zero", if you entirely disable this option you will obtain a straight jump.
In both cases not an efficient path planning.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]